EdgeTX/edgetx #6: No script for NV14 building

richardclli, 2021-05-11:
There is now script for NV14 building.  I will see if I can add one.  Please help to verify if I made any mistakes.  Thanks.

raphaelcoeffic, 2021-05-11:
@richardclli I'm waiting for @qba667 for that, as I don't have the hardware.

richardclli, 2021-05-14:
I will work out one and create pull request when ready.

raphaelcoeffic, 2021-05-14:
> I will work out one and create pull request when ready. I have the hardware.

Lovely!

richardclli, 2021-05-18:
I have added the script and creates a pull request, however the build is not success, there are some problem in the nv14 target, which is now broken due to the changes in EdgeTX, I will see if I can fix the problems anyway.

richardclli, 2021-05-18:
Just check with OpenTX 2.4, NV14 build is not working too, well anyway.  Will try to see if anything can be done.

richardclli, 2021-05-18:
Fixed a lot of problems of NV14 build, but got stuck in LUA part, can anybody try to merge my pull request and give me any advice how to resolve the build error?  I already amended nv14 definition in lua export scripts, but still no luck:

In file included from /opentx/radio/src/opentx.h:24:0,
                 from /opentx/radio/src/lua/api_general.cpp:23:
/opentx/radio/src/lua/api_general.cpp: In function 'bool luaFindFieldByName(const char*, LuaField&, unsigned int)':
/opentx/radio/src/lua/api_general.cpp:315:32: error: 'luaSingleFields' was not declared in this scope
   for (unsigned int n=0; n<DIM(luaSingleFields); ++n) {
                                ^
/opentx/radio/src/definitions.h:87:31: note: in definition of macro 'DIM'
   #define DIM(__arr) (sizeof((__arr)) / sizeof((__arr)[0]))
                               ^~~~~
/opentx/radio/src/lua/api_general.cpp:331:32: error: 'luaMultipleFields' was not declared in this scope
   for (unsigned int n=0; n<DIM(luaMultipleFields); ++n) {
                                ^
/opentx/radio/src/definitions.h:87:31: note: in definition of macro 'DIM'
   #define DIM(__arr) (sizeof((__arr)) / sizeof((__arr)[0]))
                               ^~~~~
/opentx/radio/src/lua/api_general.cpp: In function 'int luaSetSerialBaudrate(lua_State*)':
/opentx/radio/src/lua/api_general.cpp:1643:35: error: 'auxSerialSetup' was not declared in this scope
     auxSerialSetup(baudrate, false);
                                   ^
/opentx/radio/src/lua/api_general.cpp: At global scope:
/opentx/radio/src/lua/api_general.cpp:1990:39: error: 'KEY_MODEL' was not declared in this scope
   { "EVT_VIRTUAL_MENU", EVT_KEY_BREAK(KEY_MODEL) },
                                       ^~~~~~~~~
/opentx/radio/src/lua/api_general.cpp:1991:43: error: 'KEY_MODEL' was not declared in this scope
   { "EVT_VIRTUAL_MENU_LONG", EVT_KEY_LONG(KEY_MODEL) },
                                           ^~~~~~~~~
/opentx/radio/src/lua/interface.cpp: In function 'bool luaLoadMixScript(uint8_t)':
/opentx/radio/src/lua/interface.cpp:626:29: error: 'struct ModelData' has no member named 'scriptsData'; did you mean 'screenData'?
   ScriptData & sd = g_model.scriptsData[index];
                             ^~~~~~~~~~~
/opentx/radio/src/lua/interface.cpp: In function 'bool luaDoOneRunPermanentScript(event_t, int, uint32_t)':
/opentx/radio/src/lua/interface.cpp:945:31: error: 'struct ModelData' has no member named 'scriptsData'; did you mean 'screenData'?
     ScriptData & sd = g_model.scriptsData[sid.reference-SCRIPT_MIX_FIRST];
                               ^~~~~~~~~~~

raphaelcoeffic, 2021-05-18:
@richardclli I‘ll have ASAP (probably this weekend).

richardclli, 2021-05-19:
I just contact Flysky if they can donate some NV14 hardware for OpenTX developers to play around with and get a positive response.  Are you interested in this?

raphaelcoeffic, 2021-05-20:
> I just contact Flysky to see if they can donate some NV14 hardware for OpenTX developers to play around with and get a positive response. Are you interested in this?

That would help, indeed!

richardclli, 2021-05-21:
I will send you an personal email to get the your mailing address.  Hope you will have fun with the new NV14 hardware.

richardclli, 2021-05-25:
Finally can build successfully, need to test and fix other problems.  Please merge first.

richardclli, 2021-06-09:
NV14 build testing and fixing now in nv14-build branch.  Anybody are welcome to help develop and test.

eshifri, 2021-06-25:
Can you please provide the link to the new branch?
Thanks.

pfeerick, 2021-06-25:
Direct URL is https://github.com/EdgeTX/edgetx/tree/nv14-build

Otherwise, just change the branch ;)
[image]

richardclli, 2021-07-07:
The branch is now usable, need more people to test.  What is now ready:
1. EdgeTX bootloader (very cool)
2. Standard touch interface, may still have minor problems and some usability problem.
3. All switches and gimbal is working properly.

eshifri, 2021-07-07:
Does it convert models from the old (2.1 based) NV14 branch?
Thank you.

raphaelcoeffic, 2021-07-08:
> Does it convert models from the old (2.1 based) NV14 branch?
> Thank you.

We might try it out. I had removed this conversion (disabled only), but if you guys tell me the NV14 format is clean 2.1, that might be doable.

eshifri, 2021-07-08:
As far as I remember the size of the structures is the same. Some reserved bit-fields were used in NV14, but it is not a big deal if they are lost.

raphaelcoeffic, 2021-07-08:
> As far as I remember the size of the structures is the same. Some reserved bit-fields were used in NV14, but it is not a big deal if they are lost.

Is that version 218?

eshifri, 2021-07-08:
I do not know such details - it was handled by Jakob.

raphaelcoeffic, 2021-07-08:
> I do not know such details - it was handled by Jakob.

It should be the 5th byte in radio.bin, or any modelXX.bin.

eshifri, 2021-07-08:
218:
project(OpenTX)

set(VERSION_MAJOR "2")
set(VERSION_MINOR "1")
set(VERSION_REVISION "8")
set(VERSION_SUFFIX $ENV{OPENTX_VERSION_SUFFIX})

raphaelcoeffic, 2021-07-08:
> 218:
> project(OpenTX)
> 
> set(VERSION_MAJOR "2")
> set(VERSION_MINOR "1")
> set(VERSION_REVISION "8")
> set(VERSION_SUFFIX $ENV{OPENTX_VERSION_SUFFIX})

This is something different.

eshifri, 2021-07-08:
Fifth byte in modelX.bin and radio.bin is 218 (0xDA)

closed 2021-07-29 by richardclli