EdgeTX/edgetx #8: Graphical Plane/Wing/Quad Setup

trashyman, 2021-05-12:
As the Title suggests,  a Graphical Plane/Wing/Quad setup when creating a new Model with easy to understand graphics.

-Choosing Aircraft type

**Airplane**

- -Choosing number of servos/Airplane Type
- -Assigning Channels to servos
- -Choosing Switches you want to have
- -Automatic mixer generation for easy Model creation
- -Dual/Triple Rates (maybe 100/60/40?) assignable to Switch

**Wing**

- -Choosing number of servos
- -Assigning Channels to servos
- -Choosing Switches you want to have
- -Automatic mixer generation for easy Model creation
- -Dual/Triple Rates (maybe 100/60/40?) assignable to Switch

**Quad**

- -Choosing Channel-Order (TAER, AETR etc.) and Mode 1/2/3/4
- -Choosing Switches for Functions like Arming/Angle/etc.

- The Switches should all be named automatically if you create a model this way

raphaelcoeffic, 2021-05-12:
@trashyman have already tried out the wizards? I believe this already covers much of it. It is normally activated by default when you create a new model, I believe.

trashyman, 2021-05-12:
> @trashyman have already tried out the wizards? I believe this already covers much of it. It is normally activated by default when you create a new model, I believe.

you mean on OpenTX ?

raphaelcoeffic, 2021-05-13:
> > @trashyman have already tried out the wizards? I believe this already covers much of it. It is normally activated by default when you create a new model, I believe.
> 
> you mean on OpenTX ?

Yes.

lshems, 2021-05-13:
I would STRONGLY advice to not make this part of the OS itself, as the lack of predefined functional mixes is exactly what opentx makes so powerful.

If you want functional setups for different types of models, it is MUCH better to do so in LUA scripting, as with the wizards now.

The issue with the wizards is however that they help you to create it using a functional user interface asking for functional settings like which outputs for ailerons, and so on, but after creation you are back to the OpenTX userinterface again for changes, with no clue how the wizard did set up what you said it should set up.

So IF the wizard would not be a one-time use creation script, but an interface that you can use to create AND update the functional model, it becomes much more powerful.

However, as OpenTX is using different internal references to the same inputs for different models, it is currently NOT interesting to develop such a wizard. This is also linked to the issue I created :Create a common input layout for all radios #9

lshems, 2021-05-13:
You might want to take a look at the script based user interface I developed for managing predefined complex openTX models, where I DID take the effort to make these scripts radio independent. 

Look on https://www.justfly.solutions/index.php/models on what that makes possible. I'm really not posting for advertising, as I am prepared to put a lot of the effort in that solution into publicly shared scripts, IF there is a better support for this approach from the EDGE devs.

raphaelcoeffic, 2021-05-13:
> IF there is a better support for this approach from the EDGE devs.

The LUA API can really be improved, I'll second you on that one!

raphaelcoeffic, 2021-05-13:
> The issue with the wizards is however that they help you to create it using a functional user interface asking for functional settings like which outputs for ailerons, and so on, but after creation [...]

This is basically the stupidity of every wizard ever, and the biggest reason I do not like them: you get something setup, ok, and then? How do you edit it?

Eldenroot, 2021-05-13:
Helicopter add please! :)

raphaelcoeffic, 2021-05-15:
> Helicopter add please! :)

I must admit my stupidity when it comes to Helis. We will need input here (and probably help).

raphaelcoeffic, 2021-05-15:
> I would STRONGLY advice to not make this part of the OS itself, as the lack of predefined functional mixes is exactly what opentx makes so powerful.
> If you want functional setups for different types of models, it is MUCH better to do so in LUA scripting, as with the wizards now.

I would also strongly encourage to have the LUA scripts distributed separately, possibly eased by integration into the flasher, like ELRS does it (I love this part, by the way ;-)).

lshems, 2021-05-15:
Give me a Heli pilot that wants to help.

TramALot, 2021-05-15:
Basically in the USA aetr for sticks, and then at least 1 channel for a
gyro.

I would imagine somebody has a brushless motor in the tail rotor now so two
motor outputs

Swashplate mixing is beyond me..

Thanks all

jrwieland, 2021-06-21:
Heli Pilot Here (dumb enough to volunteer to help). - it took me awhile to read through everything to find this.

I can create a basic wizard using basic heli setups for both flybar and FLB.

I have helped setup many birds  on the FB forums it boils down to a couple of schools of thought.

A. Old school education taught by "Finless Bob" spool up setting with  V curves (flybar heli's).
B.  Modern ESC education (soft start) and using primarily straight curves to control the pitch & motor and a switch for killing the throttle.

The only variable that seems to be out of reach of a LUA script is the swash style [used only for flybarless. 120 (most common) 120x (blade Heli's) and  140 (rare).]
A script can set the inputs/mixes/outputs to match these setups.  However the user would have to change this one setting to ensure the cyclic mixing was correct since it is controlled through the OpenTX software.

Quick education on the cyclic's (it is nothing more than a predefined mix within OpenTX) 
Cyclic 1 - (longitude) [forward example] elevator servo up, aileron and Aux servo's down
Cyclic 2 - (Latitude) [roll right example] Aileron servo up, Aux Servo Down, Elev servo steady
Cyclic 3 - (Pitch) [Vertical example] All servos up (elev, aileron, and aux)
Mixes reference these as the source instead of the gimbal channels so that all 3 work in unison.

As far as I can tell the heli wizard was abandoned because a basic model creation (without a wizard) will fly 90% of the heli's with 2 additions in the mix page

1.  Chanel 5 Use the MAX setting and set above 50 (tail gyro) for the FBL controller to operate.
2. Channel 6 Pitch curves ( simple -100 to 100 curve will work ).

The simple model creation **will not work for inverted flight**  but to get it up and fly it (albeit piss-poorly) but it will work.
After that, it is tweaking it to the schools of thought on Throttle settings .
_______________________________________
Next item on the agenda - using  a wizard reviewer after creating - this should be possible to populate a page for reviewing items using the various "get commands".  However, I think this detracts from the and purpose flexibility of the OpenTX system ("your radio your way") and makes it seem more like every other Brand - Limited.

Eldenroot, 2021-06-21:
This would be cool!

Please check Spectrum or Jeti wizards for helis. It could help a lot and save time!

I think that some fbl units need 90 angle setting for swashplate and all mixing is done in the fbl unit itself.

jrwieland, 2021-06-21:
> 
> 
> This would be cool!
> 
> Please check Spectrum or Jeti wizards for helis. It could help a lot and save time!
> 
> I think that some fbl units need 90 angle setting for swashplate and all mixing is done in the fbl unit itself.

I have the basic setup in mind for the wizard (steps).
For any of the FLB units it doesn't matter what the swash style is since it is all done within the FBL unit. 

There would not be a need to access the heli page for them.  Only FB heli's need access to the page AFAIK. 
All of mine are 120 and this is the default on the page, .... So haven't experimented to see what happens if i change it....something to do tonight :joy: :beer:

jrwieland, 2021-06-21:
@ZZ-CAT - care to test the wizard when I get it operational - I noticed the issue you placed on the cyclics and could use your feedback for creating heli's that need Cyclic mixes.  Shouldn't take long once @raphaelcoeffic says go for it.

raphaelcoeffic, 2021-06-21:
> @ZZ-Cat - care to test the wizard when I get it operational - I noticed the issue you placed on the cyclics and could use your feedback for creating heli's that need Cyclic mixes. Shouldn't take long once @raphaelcoeffic says go for it.

Go go go! 😅

jrwieland, 2021-06-21:
## We have a green Flag....
:construction: underway

ZZ-Cat, 2021-06-21:
> @ZZ-CAT - care to test the wizard when I get it operational - I noticed the issue you placed on the cyclics and could use your feedback for creating heli's that need Cyclic mixes.  Shouldn't take long once @raphaelcoeffic says go for it.

Sure. I reckon I could do that.

In my experiences with modern flybarless flight controllers, the only time I have ever needed to use the RadioMaster's internal swashplate mixing functions was when I needed to mechanically setup the swashplate, to get everything level.

Once I got that done, I turned off the RadioMaster's swashplate mixing functions & went straight back over to standard AETR; & set the swashplate mixing up in the flybarless flight controller afterwards.

But, yea, nah. I can test it out, once ya get 'er done. =^/.^=

jrwieland, 2021-06-21:
> 
> 
> > @ZZ-Cat - care to test the wizard when I get it operational - I noticed the issue you placed on the cyclics and could use your feedback for creating heli's that need Cyclic mixes.  Shouldn't take long once @raphaelcoeffic says go for it.
> 
> Sure. I reckon I could do that.
> 
> In my experiences with modern flybarless flight controllers, the only time I have ever needed to use the RadioMaster's internal swashplate mixing functions was when I needed to mechanically setup the swashplate, to get everything level.
> 
> Once I got that done, I turned off the RadioMaster's swashplate mixing functions & went straight back over to standard AETR; & set the swashplate mixing up in the flybarless flight controller afterwards.
> 
> But, yea, nah. I can test it out, once ya get 'er done. =^/.^=
> 
> PS: I will say this as a caveat, though: The only helis that are in my little fleet, they all use 120° CCPM swashplate configurations, with all three servos directly connected to the swashplate.
> As far as I know, this is like... the de facto standard for modern helicopters, now.

OK fair enough, Having another set of eyes check what it's doing is better than just my own.  Your right, modern FBL do not need the heli tab.  I'll let you know what happens when I play around with the swash setting tonight (going to try 120X and see if it reverses my servo throws.

ZZ-Cat, 2021-06-21:
> OK fair enough, Having another set of eyes check what it's doing is better than just my own.  Your right, modern FBL do not need the heli tab.  I'll let you know what happens when I play around with [...]

Right-oh. =^/.^=

JimB40, 2021-06-21:
Guys, Wizard is 100% LUA app functionality. 
Every small change in C++ code need tests to ensure overall stability etc. plus flashing new firmware while if written in LUA it needs 5 mins to update files on SD card.
LUA API should be updated with model.getHeli() and everything is doable in LUA.

lshems, 2021-06-21:
Do we even need the Heli page? the mixing there is pretty basic to set up using LUA. and some mixer lines.

Then the swashplate limiter itself, just to prove it can be done in OpenTX without mixers or whatever.
https://www.justfly.solutions/index.php/full-repository?view=product&id=99:heli-swash-ring-limiter-mix-demo

I'm really no heli pilot at all, but was interested from the wizard perspective some time ago. If you build a wizard anyway, why not build the proper OpenTX equivalent of the helipage without ruining memory :).

(really just ignore my comments here if they are not in your line of thinking. I'm just challenging)

jrwieland, 2021-06-21:
@JimB40 I understand that's why I can assist in making one. 
My C coding skills are over 35 years old.  LUA on the other hand is good.

Minor tweaks using the existing multi-rotor setup to create one.
Biggest part is creating the graphics for the interface.
Steps for creation are about the same. 
Trying not to re-create the world just help where possible

jrwieland, 2021-06-21:
> Do we even need the Heli page? the mixing there is pretty basic to set up using LUA. and some mixer lines.
>
> Then the swashplate limiter itself, just to prove it can be done in OpenTX without mixers [...]

@Ishems The only ones that really "need" or would use  a wizard are the remaining FB heli guys coming to OTX.
The setup curve to get the mixes right is severe.  Trying to coordinate the 3 servo movements without maxing them out at the extreme throws is a lot of fine tuning in the mixes.

I'm not opposed to writing the lua.  You have seen on FB and the RC forums that it gets asked for ever so often.  Usually from those coming over to OTX.

jrwieland, 2021-06-23:
well that was a fun exercise, but unachievable in the current Lua settings.
There is no access to the 5 settings on the heli page.
we can manipulate the sensors, inputs, outputs, mixes and curves in the avail lua 2.3 commands. but access to any of the 
swash setttings
swashType
swashRingVal
swashElevator Source/Weight
swashAileron Source/Weight or swashCollectiveWeight

lshems, 2021-06-23:
As I said, I can finish it with a curve and a few mixes.

What do you have so far, post it, I'll see if I can add that.

The settings adjustable I will label them on inputs .

Let's get this done without the Heli page.

jrwieland, 2021-06-23:
it can be done without the heli page but the internal mixes would add several mix lines without the cyclic mixes.  It is more than limiting the swash movement.   For example to set up a quad for a singe flight mode without it's controller doing the mixes, it would take 16 mixes to control it (4 motor adjustments per gimbal axis).  A heli needs 3 servo's to work together on each axis.  an additional dummy mix line would have to be added to control the servo on channel 3 without the cyclics.

Eldenroot, 2021-06-23:
@raphaelcoeffic - is possible to add a support for missing features?

lshems, 2021-06-23:
But the Heli page is not fixing that for you.

The Heli page does nothing more than mixing the collective pitch into the roll and pitch, converted by simple geometrics to the swash servos, 3 of them. With the roll and pitch being on one stick and limited in out to the swash plate diameter.

Quads is another story, but the Heli page doesn't help.

lshems, 2021-06-23:
The swash angles derermine the fractions to be mixed, easily set by Lua.

lshems, 2021-06-23:
And we have 64 mixer lines. Isn't a wizard meant to support quick setup of tedious stuff and open possibilities that are practically undoable 'manually'?

raphaelcoeffic, 2021-06-23:
> @raphaelcoeffic - is possible to add a support for missing features?

What exactly should be added / is missing?

lshems, 2021-06-23:
I've been messing around it some time ago.

I think most people agree that in a basic Heli setup (not using a flight controller of any kind) it would be very practical to have flight modes for switching the pitch modes. The last addition to opentx has a serious bug in trims in adapting flightmodes through LUA (to set up a switch to the flightmodes). That bug is described here already.

Then, either the basic parameters of the Heli page should be accessible for the swashplate mixes for different head types, or someone has the guts to do it with mixers and a curve only. It is work, but I still think basic arithmetics. So not an undoable job for a Lua widget.

jrwieland, 2021-06-23:
@lshems What your missing is the cyc1 2 & 3  source is set on the heli page - it eliminates the needs for additional lines to be be added for the manual mixing.  Making it easier to get something setup.  Isn't that the purpose of the software to make things easier?  We can go back to the 70's and manually mix everything, a lot more trial and error but we can do it.  I have looked at your code for the limiting the swash angles.  I think your missing the basic fundamentals of what the heli page is doing for the flybar'd heli's.  

Modern FBL heli's do not need the heli page at all. They are no different to set up than a quad or simple plane.

I really don't want to get into a pissing match over this so, why don't you create a wizard and we'll see what works, perhaps I am wrong and making too much of it.  So I'll continue on my path, you create one on your train of thought and we'll see what works.

As far as changing the curves via flight modes, this is easily done with a mixing.  Set curve 1 to fm1 and curve 2 to fm 2-.... it is no different than the throttle curves for (curve 1 for throttle up or sport flying throttle 2 for light 3d and throttle 3 for full 3d).  I'm not saying that there isn't a bug between the fm and LUA. it is documented.  But I have no need for the script so I haven't followed it at all.

lshems, 2021-06-23:
I'm really not trying to be the smart guy here. Just challenging the need of the Heli page. And you are right, I should prove that by example.

Fun job for this evening, and really, shoot the result if it is not ok. 

Can you explain how the Heli page would be if use for multirotors? Do you have an example setup?

Would help if I could understand that.

jrwieland, 2021-06-23:
> 
> 
> > @raphaelcoeffic - is possible to add a support for missing features?
> 
> What exactly should be added / is missing?

The sources for the Cyclics are set on the heli page - from reading the simulator code they are:

swashType (sets the geometry of the swash)
swashRingVal (usually left a zero)
swashElevator Source/Weight (Source for Cyc1)
swashAileron Source/Weight and Source for (Cyc2)
swashCollective Source/Weight Source for (Cyc3)

for the majority of the FB heli's out there the default setting to the sources would be 
Swash type 120
Long (cyc1)  - Elevator Stick
Lat (cyc2) - Ailer Stick
Collective  - Throttle Stick

jrwieland, 2021-06-23:
@lshems There isn't a need for the heli page for anything except the old flybar helicopters.
If they all disappear then there would absolutely no need for the page at all.

Flybar Helicopters only have a receiver.  (no controller)
The gyro only controls the tail servo it has nothing to do with the swash at all.

FBL helicopters have both a receiver and controller (same as a quad).  They do all the mixing internally and there is no need to do any mixing or limiting of the swash.  That is performed with the (BeastX, Brain, Vbar) Controller when setting it up.  These heli's do not need a heli page (but some still use it as a check against their controller and then disable it after they are satisfied).

ZZ-Cat, 2021-06-23:
> @lshems There isn't a need for the heli page for anything except the old flybar helicopters.
> If they all disappear then there would absolutely no need for the page at all.
>
> Flybar Helicopters only [...]

To be fair, even on my T-Rex 500 ESP 3G (which has a BeastX Micro Beast Plus flight controller), the only time I've ever used that Heli Page is to aid in setting up the servos & mechanical geometry around the swashplate & aiding the setup of the BeastX unit, when it comes to centering the servos. Other than that, I never needed to use it afterwards.

jrwieland, 2021-06-24:
I have 1 flybar heli ( an old trex 450 pro) that is the only heli that accesses the cyc inputs.  The remainder are are all fbl, Goblin 500, trex 700 dfc ... on and on....(addiction is fun)  and they do not use the cyc inputs. 

In creating a heli wizard, it needs to address both types, or else it is only a fbl setup and the old school boys who we all (multi-rotor and heli flyers) owe our hobby innovations to; for their initial creation and upgrading of the original gyro, will be on their own again.  

It was from that tail gyro foundation that the 3 axis and subsequent 6 axis controllers are based off of.  So to say that they do not matter anymore is the same as the current push to remove all of the past history and only show the present.  Without the past and building off of it, there is only the mindless hampster wheel mentality (Running in circles without going anywhere or without a definitive goal in mind). But that is a discussion for another time/place.  This is for innovation and moving forward.

ZZ-Cat, 2021-06-24:
> I have 1 flybar heli ( an old trex 450 pro) that is the only heli that accesses the cyc inputs. The remainder are are all fbl, Goblin 500, trex 700 dfc ... on and on....(addiction is fun) and they [...]

I agree. It actually is a good idea to be accommodating of both flybarred & flybarless machines. =^/.^=

lshems, 2021-06-24:
Just to prove it can be done.

9 mixers for the cycle
9 for the swashplate limiter

The swashplate value can be set as weight on an input.
The other weights with lua.

So a wizard can also be done.

[Helipage in mixers.zip](https://github.com/EdgeTX/edgetx/files/6705852/Helipage.in.mixers.zip)

jrwieland, 2021-06-24:
@lshems I never said it couldn't be done.  But that may mixes for what could be accomplished with fewer is why there is cyc1, 2, 3.  Your example is how it was done before flight controllers.

Here is a flybar 450 model with 3 flight modes for reference,
 it much easier for the user to change a setting.
[Trex 450 MOD.zip](https://github.com/EdgeTX/edgetx/files/6706010/Trex.450.MOD.zip)

pfeerick, 2021-06-24:
He he... I was going to say "lets just say...  it's *possible* ... but it's not _practical_" ... be cause... seriously... 18 mixes instead a handful of settings... but, I think the point lshems is trying to make here is that the mixes themselves can be done in the main mixes, meaning you could ditch the heli setup screen completely, and replace it with a lua script... one less thing in the main transmitter firmware, can be more easily updated and customized in the future.

JimB40, 2021-06-24:
Trying to follow but HELIs it terra incognita for me. :)

My point of view (UX) is that if something can be archived by configutation it should be done via App (wizard or whatever we call it). INPUTS, MIXES, OUTPUTS are general while HELI seems to be funcional. You got the knowledge to answer.

Another thing is we review now OS dumping unnecessary legacy stuff. So if this is for 1% cases and can be done in other way we should consider dumping HELI tab. There is always OTX for legacy solutions. :)

jrwieland, 2021-06-24:
true, OTX can be used for the legacy, and if the dev's general consensus is to drop the heli tab then all of this is mute.  

I won't go any further with the creation of the wizard since there is going to be no support for the FB heli's (via cyc1-3),  this was the reason one had not been created.  Modern electronic mixing (FBL) are fairly easy to setup without a wizard, you only have to add a single mix  to ch5 and 6 (gyro and pitch respectively)  to any model created without a wizard.   The legacy cyc1, 2, 3 inputs can also be eliminated from the code since it could be accessed from OTX.   

The graphical plane, multi rotor and gliders all have wizards.

I don't believe there are any other open issues with this item, so I suggest closing it and moving on with stripping the legacy items from the code.

lshems, 2021-06-24:
A cyclic mix is nothing else then an arithmetic excersize to convert one frame of reference with 3 inputs roll, pitch and throttle to another frame of reference, the swash plate orientation, with three servos being the geometrical connection. The only extra input needed is the geometric setup of the swash (90, 120, ....).

Issue is that with a mixer line being capable of capturing only one input, so needing three inputs to create one new output, you get 9 mixerlines, where you only have 1 factor defining all that. 

The Heli page solves that partially by NOT specifying the outputs, but using the predefined cycl1,2 and 3. This allows to limit the page to the three inputs. The weights are actually not needed, as you could stage an input for the raw stick input as in all other mixes.

The swash limiter is a function to limit the output of one of the resulting output references, being the 'tilt' of the swashplate, to one value, where without it, it will have different maximum values that do not make sense in the new reference system, in the corners of the gimbal positions. This takes currently 9 mixerlines to execute a squaring of the X,y movements, add them, and do a square root operation by using a curve as a lookup on the result. (Nice).

Those are two different things that could be tackled in different ways. A Lua mixerscript could do the trick, but is considered unreliable.

lshems, 2021-06-24:
A Lua mixerscript is actually doing exactly that: taking three (4) inputs an creating 3 (4) outputs

lshems, 2021-06-24:
Could we create an 'internal' Lua mixerscript to choose from that IS reliable?

lshems, 2021-06-24:
Only for cyclic mixing, on for limiting X,y coordinates to a maximum radius.

jrwieland, 2021-06-24:
there is no point, by stripping the code it of the legacy items Edge becomes the cutting edge for multi rotors and planes. Heli pilots can still use it however there is no software support.  In creating a wizard or any lua it is implying that helicopters (all) are supported and this will not be the case.

lshems, 2021-06-24:
That would eliminate the need for that page completely. And the infrastructure is there.

It would be a very nice way to support these kind of special mixes.

lshems, 2021-06-24:
I can think of others as well: flaperon mix, safety switch mix, ....

jrwieland, 2021-06-24:
the page could always be turned off by checking the noheli box on the build version, same as GV 's and nooverridech.  There was also the choice to turn on modules (bluetooth, internal GPS etc.) allowing the user to choose if they wanted that support.  In moving forward and eliminating the legacy items why include a lua code.  I already wipe out all of the unnecessary lua codes that cater to muti-rotors and gliders when i download a version.  Ghost and crossfire for example are of no value to me, and take up unnecessary space on my sd card.

lshems, 2021-06-24:
That's not the point I think.

jrwieland, 2021-06-24:
Leaving this to the dev's. if they decide to eliminate it, then there is nothing more to to with this item.  if it remains then I'll come back on board and work on tis otherwise, I am moving on and looking for other items to assist on. 
:v:  :running_man: :dash:

lshems, 2021-06-24:
What I hope is that this can be picked up in a less intrusive way than build options, for actually two simple mixes that don't fit in the regular mixing strategy. (And thus need an extra page now)

We have the (Lua) mixer functions for that, where a screen is offered comparable to the normal mixer line dialog boxes for entry of the relevant parameters, and the outputs are defined by the mixer function called.

This should be the general way of creating special mixers. It is designed for it.

Only issue is apparently the reliability.

Lots of reasons to doubt reliability, but if we could work on trust on the reliability, you get endless possibilities in return, including the EXACT same functionality as the Heli page, and countless others.

I think it is in the line of thought for EDGE to investigate the possibility to get the mixer scripts up to a reliability that makes them fit for flying functions. I personally think they are already, when used with consideration.

One possible way us to include 'fit for flying mixerscripts as integrated c mixers, so that they are not dependant on the Lua environment.

Just some thoughts.

P.s. I sense a bit of disappointment in the conversation. It was by no way my intention to block any developments based on the Heli page. I just think it should be challenged, as I argue there are better ways to do it (not referring to my proof of doing it with opentx without Heli page).

And my arguments are only arguments, to be weighed by the community on validity and weight.

So sorry if I might have offended someone or gave the impression I'm not in favour of progress, and would want to eradicate the Heli page. I have no opinion about leaving it in or out, as I don't have to maintain it, and it is not in my way.

pfeerick, 2021-06-24:
I with your points on moving it to lua @lshems ... whilst it could be seen as just shuffling things around, there is no denying it would make the build process that bit easier as it would no longer need to be in the main code, and thus doesn't need to be build time option. If you want the heli config, just copy the lua wizard onto your SD card, if you don't want it, delete it. No need to go through and have a new firmware built and uploaded. Having it in lua means more people can work on it, and customise it. 

I see not reason for ETX to leave heli's behind - saying that there's always OTX to me feels we're evading the issue, rather than resolving it one way or another.

lshems, 2021-06-25:
I have opened another issue on the assumed lack of reliabiliy for mixer scripts for in flight functions.

There needs to be some attention to that, as moving those two mixers to Lua mixers is no issue at all, but the reliability is.

And Lua access to set a Lua mixer is needed if you want to use it in a wizard.

I suggest to not close this issue,but use it for the wizard.

We need API access to the mixer scripts, and a wizard is possible if Lua mixers are safe to use in flight.

So a lot of IFS.

And I do agree that Heli support is needed for any serious radio, even if seldomly used. It's like a marriage without rings. The rings don't do anything, but without ....

lshems, 2021-06-25:
Writing the two Lua mixer scripts is really not a big deal. I'll fix that in two separate issues as well.

[2 more replies]