EdgeTX/edgetx #11: Please add T18 / TX18S build

Ender1967, 2021-05-13:
There is no firmware build ATM for T18/TX18S, please consider adding these models

**Additional context**
Add any other context or screenshots about the feature request here.

raphaelcoeffic, 2021-05-13:
> There is no firmware build ATM for T18/TX18S, please consider adding these models.
> some models of T18 /TX18s have touch screen, some models don't... maybe it will be better to divide in two models

Sure! Luckily, I'm getting one to test in the next days. I was hoping to be able to detect the touch chip (I2C) and take it from there, without to really have to split it into multiple targets.

sfjuocekr, 2021-05-16:
I was working on the RadioKing TX18s for myself mostly here a while ago: https://github.com/sfjuocekr/opentx/commits/2.3-tx18s

You can just use the Jumper T18 build, it is mostly just a cosmetic thing and something with the screen brightness or shared pin I have not investigated further that makes the hiss on the speaker.

RadioKing is not very forthcoming in their e-mail responses, the last contact I had with them was that they would put this through to their "tech team".

I did not investigate the touchscreen any further than it does not work with the current LCD touch controller (it does not show up in the log or initialize).

raphaelcoeffic, 2021-05-16:
@sfjuocekr thx a lot! Can you make it against the 2.4 branch please?

sfjuocekr, 2021-05-16:
My current system (21.04) won't build, at least not the way I used to before.

Once I figure out how to make a working build I'll be back on it :)

Maybe you can give a short synopsis on how to get a build started? What I did in the past was make a build directory and run cmake -DPCB=X10 -PCBREV=T18 etc ...

raphaelcoeffic, 2021-05-17:
@sfjuocekr can you please join dev channel on our discord server? You will get all the help you need.

sfjuocekr, 2021-05-17:
I did last night, I think I've got a build going and will test it later today!

richardclli, 2021-05-20:
You can use the tools/build-jumper.py -b T18 -t EN

sfjuocekr, 2021-05-21:
That has never worked for me:

```
../tools/build-jumper.py -b T18 -t EN ..
cmake -DPCB=X10 -DPCBREV=T18 -DINTERNAL_MODULE_MULTI=YES -DDEFAULT_MODE=2 -DTRANSLATIONS=EN -DJUMPER_RELEASE=YES ..
**CMake Error: The source directory "/tmp" does not appear to contain CMakeLists.txt.**
```

Running cmake manually always works for me

raphaelcoeffic, 2021-05-22:
@sfjuocekr With your compile fixes, I get the T18 to boot into bootloader mode, however it still seems to crash somewhere early when booting the main app, no clue why/where now. It does not seem to be related to the NRST line of the screen, as the radio would otherwise still work, even if the screen is unduly reset (I think, at least).

sfjuocekr, 2021-05-27:
Ah I had the same issue with my previous builds and fixed that by changing the toolchain, it probably locks up right before it should say "Welcome to OpenTX" right?

My mom had the bright idea to take me away for vacation, so I have not had the time to do any testing. I just got home and have some stuff to fix here first.

rotorman, 2021-05-27:
Very interested to learn which toolchain did you opt to using, as I am having bad results with anything above 6 2017-q2 when LUA is included in the build.

sfjuocekr, 2021-05-27:
gcc-arm-none-eabi-4_7-2013q3-20130916-linux

rotorman, 2021-05-27:
Whoa, that's old... and is the default of OpenTX 2.3.
Are you saying it was not working with ETX default 6 2017 q2 for you? Which OS are you using for building?

sfjuocekr, 2021-05-27:
Yes on Linux, this was a while back when I tried to compile my own builds and could only get it to work with that version.

I did not investigate why the newer toolchains did not work, at first I just pulled the one that came with 20.10 but I'm now on 21.04 on my desktop.

rotorman, 2021-05-27:
Could you try to make it work with 6 2017-q2 (as EdgeTX default)?

sfjuocekr, 2021-05-27:
If I find the time, sure. But my I have lots of birthdays this week and WoW TBC is released on my own birthday as well :(

I'll probably not be able to do much for the coming month.

raphaelcoeffic, 2021-05-27:
@sfjuocekr are you saying the T18 starts when the 2.4 branch is compiled with GCC 4.7? It does not even compile here with that old GCC. Really, for most targets this is the opposite here: works perfectly with gcc 6, but does not start otherwise.

raphaelcoeffic, 2021-05-29:
Ok, T18 build has been added and seems functional to me, closing now.

closed 2021-05-29 by raphaelcoeffic

sfjuocekr, 2021-07-09:
I can confirm the T18 builds work on the TX18s.

It doesn't even hiss like it used to!