EdgeTX/edgetx #9: Create a common input layout for all radios

lshems, 2021-05-13:
Currently, every radio has a different number of switches defined, trims, etc.  

This is of course defined by hardware available.

Yet, if there would be one virtualization of all radios, where you can map all physical switches to the common radio model, and assign physical inputs or fixed values of up,mid and down, or even percentage values, to all those inputs at will, all models and scripts would become interchangeable between radio models.

We need this to increase interoperability of models and scripts, so we can finally start community effort on sharing these.

See https://www.justfly.solutions/index.php/full-repository

raphaelcoeffic, 2021-05-13:
I’m not sure I understand properly. But there is something you might like. With the new storage mode based on YAML, we should reach a state where models can be seamlessly exchanged between radios. So of course, it will only work with switches & pots available on both radios.

lshems, 2021-05-13:
Ok, stay with me:

I currently create an input in the inputs page for ANY functional control input for the model that I might want link to a switch. This input is then used throughout the model, instead of the direct input from the hardware.

Depending on available hardware switches and trims, I can then assign to this functional control input any physical switch or input I like. If I don't have enough physical switches for all functions, I have to choose to put some functional inputs in a 'fixed' state by assigning the MAX as a source, and play with the weight to get the wanted default functional behavior for that function control input to the model.

However,  actual physical switches are part of the model definition now, which prevents for instance a model exchange in the field by switching SDcards and restoring a backup model from one radio type to another.

If this concept of assigning fixed values to 'physical' switches that are not there for radio x would be introduced in the calibration and hardware assignment page, together with the possibility to set positive AND negative values for the endpoints in the calibration, you would get complete freedom in assigning ANY physical layout to the standard model layout, including out of the box support for inverting switch behavior.

Lengthy post, but it boils down to putting the physical available inputs in a radio firmware controlled list, and allow mapping to a general model definition that should contain for instance 64 slots for physical inputs, that can be used either as switch or analog input, depening on the hardware inout you assign to it.

You could even assign sliders to model switch inputs and vice versa. Have it populated with a default setup by having the model ask to the firmware for the entries of the gimbals, trims and the two primary switches, and leave the rest to the users, and you have complete interoperability between physically different radios, because the exchanged model info doesnt contain the radio specific mapping. That is in the firmware!

raphaelcoeffic, 2021-05-13:
Ok, I think I get the concept, thx for explaining. This will require quite some changes, but I'm open to it once the transition to YAML is done. That will give us much more flexibility in terms of internal data model, which is not possible right now without huge efforts. But frankly, yes, I like the idea. And I know your history with the OTX team very well, believe me ;-)

lshems, 2021-05-13:
I'm patient, but would love to see things that REALLY open up for low threshold community effort sharing

lshems, 2025-11-06:
Is this something now more viable?

pfeerick, 2025-11-09:
This would be nice to have... as mapping of physical inputs (switches, pots, sliders) to inputs used in models is something that would greatly help when moving between radios. I've been thinking something of something along the lines of naming the purpose of the switches, and then they can be dynamically assigned as the model is loaded... i.e. if you configure a switch as "arm", and a model file is looking for "arm" switch, it would automagically find it when loading the model (once you've configured "arm" to be a specific control)... or fall back to a low value (i.e. low if not defined). 

Is this how this would be expected to functionally work? i.e. a new input class/group, that would map physical controls to the "reassignable" inputs/controls in radio setup? Which would apply to both sources and switches (thus can be used _everywhere_).