EdgeTX/edgetx #13: Touch interface - Radio Settings - Make Pwr Off delay a listbox

pfeerick, 2021-05-14:
It looks like the input for 'Pwr Off delay' has a limited range of options - 0s, 1s, 2s and 3s - or at least it does on the Radiomaster TX-16s firmware. Making it a list like say 'Backlight Mode' would make it more accessible for touch.

raphaelcoeffic, 2021-05-14:
> It looks like the input for 'Pwr Off delay' has a limited range of options - 0s, 1s, 2s and 3s - or at least it does on the Radiomaster TX-16s firmware. Making it a list like say 'Backlight Mode' would make it more accessible for touch.

Thx for feedback / input. There are more of these. Luckily, it should en easy to convert these fields into drop-downs.

pfeerick, 2021-05-16:
I've tweaked the title, and added more fields I think are worth _considering_ for listbox entries... it's not a list of demands! 😁

philmoz, 2023-01-18:
Many of these would result in very long lists, so I'm not sure they would be good candidates:

Date - year: current range is 2018 - 2100 (83 entries)
Time - minute & second - 60 entries

I think it would look odd to have some parts of the date as numbers and some as lists. I would leave them all as numbers.

Vario - pitch zero - 81 entries, pitch max - 141 entries, repeat zero - 81 entries.
Alarm - battery low - 91 entries.
Play delay - 101 entries.

All of these would he hard to navigate as lists.

I don't have GPS in my radio; but I'm guessing 24 entries. Could go either way.

Power off delay has already been converted.

gagarinlg, 2023-01-18:
[image]
This might work for the time entry, even without touch, when implemented right.

For the year, the list could be shortened.

For the others, you are right in my opinion.

pfeerick, 2023-01-18:
Given the plans to overhaul the UI nearly entirely, this issue probably no longer has a place anymore anyway. Thus the decision (and blame) on what controls are best for each input are in hands of the guys working on the UI :grin:

closed 2023-01-18 by pfeerick