EdgeTX/edgetx #17: Touch interface - back button

pfeerick, 2021-05-15:
The back button for a lot of menus is currently overlaid over the logo in the right corner... sort of. 

[image]

I actually think it should be kept where it is, but make it bigger and blank that logo off completely. Then it becomes "press the logo" (and/or top row) of the screen for options, and then when in the menus press that same spot to go back.

raphaelcoeffic, 2021-05-15:
I was think to actually make the "back button" disappear, and have the "opentx logo" be the "home / back" button. The hitbox is a bit small now, so I'll extend it to cover the complete logo, so it is easier to "click" on it.

pfeerick, 2021-05-15:
I like the idea, but putting myself in the shoes of someone who isn't familiar with that behavior yet (i.e. new user), I would have to ask what cue/clue is there that pressing the logo would take me back to the previous screen/menu? i.e. how can it be made intuitive to use?

raphaelcoeffic, 2021-05-15:
Very good question! I have no answer right now.

raphaelcoeffic, 2021-05-15:
Looks like we are heading towards a "Home" button. It should look like a "Home" button obviously. Artwork is welcome :-)

lshems, 2021-05-17:
For any touch interface, menu structures are the main underlying framework, together with a popup keyboard for numbers or letters to enter data, if not ranged.

So I would think any attribute needs to be findable by clicking only, and then either show a popup list, number keyboard or letter keyboard.

Just thinking.

pfeerick, 2021-08-21:
This was basically resolved between https://github.com/EdgeTX/edgetx/pull/533 and https://github.com/EdgeTX/edgetx/pull/584

There isn't a back 'icon' as such, but the screen that could not be backed out of via touch can now be backed out of by pressing the EdgeTX logo, thus consistent with the rest of the UX.

closed 2021-08-21 by pfeerick