EdgeTX/edgetx #18: Bug: Touch Screen - Top bar setup widgets - can't get back

pfeerick, 2021-05-15:
Earning my keep, breaking stuff...

1. Top bar
2. Screens setup
3. Setup widgets
4. Select any of four boxes
5. Select widget
6. Select any widget
7. Press some other part of of the screen other than a top bar widget
8. Press return until the cows come home with no luck
9. Press on one of the top bar widget boxes
10. Press away from the options
11. Pressing return will now work

There _may_ be a second state related bug...  At step 3, you can still get into any of other menus by pressing the top bar (other than a widget box), meaning you can go into the menus whilst in edit mode... 

TX16s

pfeerick, 2021-05-19:
Eirik "grep boy" Toft on the discord did a video showing this: https://youtu.be/t6h5e1KAylk

raphaelcoeffic, 2021-05-19:
I think the "RTN" (weird name when actually "EXIT" is meant.....) button should do that, but many more input is allowed whereby it should not. The fix looks like this: forbid other inputs so the UI does not get input that bugs state.

pfeerick, 2021-05-20:
Part of that should since you are editing the top widgets, to lock out touch input outside of the top bar ;)

closed 2021-05-22 by raphaelcoeffic

pfeerick, 2021-05-22:
I can confirm... no more escaping from the widget setup! :)

commit 61abc465 (2021-05-22): Catch gestures outside widgets to be setup

resolves #18