EdgeTX/edgetx #19: Mixer / Output mapping

raphaelcoeffic, 2021-05-15:
Miami Mike described it pretty well in his blog on RCG:
https://www.rcgroups.com/forums/showthread.php?3883631-How-OpenTX-should-have-worked

RC-SOAR, 2021-05-16:
The proposal adds a layer of indirection between mixers and outputs. It could lead to 

- 'orphan' mixers which are not linked to any output
- individual mixes being linked to multiple outputs.

This will have a major impact on ease of debugging via the Monitors screen, since the latter depends on the 1:1 relationship between mixes and outputs.  

An easier approach to the problem is to simply have up/down buttons on the Outputs, to allow re-ordering.

raphaelcoeffic, 2021-05-16:
@RC-SOAR I‘m not sure I understand what you’re proposing. Would you be so kind to elaborate a bit on that?

IMHO, the key idea here is to have some possibility to go beyond this 1:1 mapping. Being on the UI level, or just “under the hood” (re-ordering requires some kind of mapping, internally).

Some have tackled it by offering a channel re-mapping in the receivers (access rx do that, don’t they?), which is a requirement I can totally understand, but is wrongly placed in the receiver, I think.

Some other manufacturers offer something closer to Mike’s original proposal.

but all in all, this is probably about having some sort of flexibility between the logical channels (mixers), and physical (output).

RC-SOAR, 2021-05-16:
@raphaelcoeffic First, thank you for the work you're doing on this. It's great to see OpenTX being further developed in this fork.

I accept the need for more flexibility, similar to what's achieved in the MPM. There are two ways it can be achieved. 

(1) Add a mapping layer as per MM's proposal, whereby an output can reference any mixer. There will no longer be a 1:1 relationship between mixers and outputs.

Or (2) allow channels to be easily re-ordered. This will maintain the 1:1 relationship.

There are a couple of issues with the re-mapping proposal. First, there may be orpaned mixers (mixers which are not referenced in any output), so they won't appear in the outputs monitors screen - even though they may be key mixes. Second, the link within each pair of bars will be broken, making it more difficult to interpret what's going on. Here's a screenshot of the monitor screen as it currently appears. It will no longer be usable in its current (well-proven) form.

[image]

Instead, I propose a simple 'move Up/Down' option in the Outputs menu.  Essentially a built in channel changer.

This will achieve the primary objective, which is to re-order the outputs to match the user's setup. Crucially, it will allow the Monitors screen to be retained. It will also work well with a touch interface. And it will not require any change in workflow or the printed reports.

More detail: with each move, 

1. The outputs object is moved up or down one slot, switching places with the adjacent output. 
2. All the related mixers are moved with the output, so the change is reflected in the Mixers menu. 

See simulation below.

[image]
[image]
[image]
[image]
[image]

RC-SOAR, 2021-05-16:
If a mapping layer is the only option available, then I guess it could be made to work by enforcing a 1:1 relationship between input and outputs. This is how the MPM works. This would avoid the possibility of orphan mixers, and you would allow existing monitor screen to be used (with some changes). 
But I still prefer the second option.

raphaelcoeffic, 2021-05-16:
Thx a lot! This is feature design! I need to digest the info 😳

jfrickmann, 2021-07-02:
I fully support this!! Whether we keep a 1:1 mapping or the original proposal is good with me. Much smarter than hacking Lua scripts that move entire mixer groups around like I did here: https://github.com/jfrickmann/SoarOTX/blob/master/SD-CARD/SCRIPTS/TELEMETRY/JF/CHANNELS.lua

lshems, 2021-07-02:
Yes, go.

But do the inputs as well. Shuffle physical switches over predefined names.