EdgeTX/edgetx #22: Bug: Selecting PPM crashes *something* on TX16s

pfeerick, 2021-05-16:
When changing External RF from `XJT` to `PPM`, the power LED and screen backlight go out, then the power led and backlight comes back on with the screen blank. Pressing 'RTN' etc appears to do nothing. When pressing and holding the power butting the power delay squares come on, and the Tx turns off. This is all with no module fitted. Equipping a Frsky XJT module does not make any observable difference. 

Does not occur when changing from `OFF` to `PPM`, and does not appear to happen when going to any other Mode. It also does not happen when going from XJT to any other mode (ie GHST, DSM2). But if you then change from that intemediate mode to PPM, it will happen.

In other words

`OFF -> PPM -> XJT` ok
`OFF -> XJT -> PPM` bad
`OFF -> XJT -> GHST` ok
`OFF -> XJT -> GHST -> PPM` bad
`OFF -> PPM -> XJT - > DSM2 -> PPM` bad

It could be a red herring, but I do not believe this to be new, but related to a [preexisting OpenTX bug which affected the 9XR-PRO](https://github.com/opentx/opentx/issues/7594), but I had not noticed on the TX16s as yet. 

tx16s-edgetx-60726891, #85ebb168

raphaelcoeffic, 2021-05-16:
This seems to be fixed with latest 2.4 (after merging some 2.3 changes). Can you please re-check?

pfeerick, 2021-05-18:
I can confirm that upon testing against 0939d1bdffb9590435aa7e51640c93b76d13a7e5 it is indeed gone. 🎉

closed 2021-05-18 by pfeerick