EdgeTX/edgetx #23: Enhancement: Bootloader progress bar

pfeerick, 2021-05-16:
The bootloader screen is dark, and the outline for the progress bar is black... basically hard to see on a colour_lcd screen TX. Maybe the outline could be a different colour? Grey, or maybe even white, given there is other white text?

raphaelcoeffic, 2021-05-16:
@pfeerick What you are looking for is here: `radio/src/targets/horus/bootloader/boot_menu.cpp`. The strange thing is however, it should already be red/green:

```
        if (st == ST_FLASHING || st == ST_FLASH_DONE) {

            LcdFlags color = BL_RED;

            if (st == ST_FLASH_DONE) {
                color = BL_GREEN;
                opt   = 100; // Completed > 100%
            }

            lcd->drawRect(70, 120, 340, 31, 2);
            lcd->drawSolidFilledRect(74, 124, (332 * opt) / 100, 23, color);
        }
```

pfeerick, 2021-05-16:
Ah, thank you... I'll try and get in a position to build tomorrow, and play with it... The bar is certainly red whilst running, and green when finished... I'm referring to the outline around it... Probably this line

https://github.com/EdgeTX/edgetx/blob/ea616c71876843a32352eab4c68d27a76ca19405/radio/src/targets/horus/bootloader/boot_menu.cpp#L144

raphaelcoeffic, 2021-05-16:
You're right, the function should have an extra param with the color. If the param is absent, it is black.

pfeerick, 2021-05-16:
Ah ha... thank you... Got it all figured out now... 
```
lcd->drawRect(70, 120, 340, 31, 2, 255, BL_FOREGROUND); //foreground white
lcd->drawRect(70, 120, 340, 31, 2, 255, BL_SELECTED); //blue used in selection
```

raphaelcoeffic, 2021-05-16:
> Ah ha... thank you... Got it all figured out now...

I took the liberty of editing your comment to use the proper constant ;-)

raphaelcoeffic, 2021-05-16:
PR please!

pfeerick, 2021-05-16:
I thought we were going to wait for feedback first? :-P And thank you for the edit... nice, so solid was an option! ;) I'll do a PR against the selection colour, and you can adopt/comment as you see fit...

PR #25 (2021-05-16): Use the selection colour as the progress bar border
Resolves #23.
closed 2021-05-16 by raphaelcoeffic