EdgeTX/edgetx #24: Cool ideas for SPI flash on X10 derivates?

raphaelcoeffic, 2021-05-16:
Many have wondered why the SPI flash is not used in OTX/ETX. IMHO, it is more for the lack of nice ideas than anything else. As pointed out here, most things are "prepared", but not used at all:
https://www.rcgroups.com/forums/showthread.php?3859585-Blog-14-RadioMaster-TX16S-SPI-Flash-readout

So now, what could we make out of it? Does it even make sense to try it out?

brainbubblersbest, 2021-06-09:
Isnt the data störend on the sd because the x12s Developer Edition not had internal storage available?
The Code was writen then to be on the sd card.
And as the final x12s came out, it surprisely had internal space. But Was not used at all.
Eg the TX16S have massive Problems with factory SD Cards.
Dont know if all Horus based Radios have internal Flash. But If so, is anything speaking against using it?

raphaelcoeffic, 2021-06-09:
@brainbubblersbest that is definitely an option. It makes things however harder when the storage is not directly available such as with an additional card.

CoderElectronics, 2021-06-09:
Stick blackbox anyone? Possibly like logging telemetry and channels sort of similar to the BF blackbox implementation.

brainbubblersbest, 2021-06-09:
> @brainbubblersbest that is definitely an option. It makes things however harder when the storage is not directly available such as with an sd card. You need extra code for filling and reading the flash.

Dual Boot Option would be another idea.

For fast testing purposes. Maybe available with a Dual Boot Compile Option that enables the Boot Manager always come before the OS. 
For safety purposes: ¿ Preventing access from etx installed on internal drive to external SD card to avoid messing the second drive data. 
This would give a little extra speed also in testing New nightlys. In the same time have a stable Release for daily use installed. Or just for comparing two branches.

gagarinlg, 2025-01-24:
Since new radios won't come with this kind of extra storage, we should close this

closed 2025-01-26 by raphaelcoeffic