EdgeTX/edgetx #27: LUA touch API

raphaelcoeffic, 2021-05-16:
As 2.4 brings touch screen support, we need to come up with an API design to allow for touch screen features to be supported in LUA scripts as well, thus allowing a lot of new features based on scripts.

First of all, we need requirements, then we can figure out how to implement them.

lshems, 2021-05-16:
First quick response: it's not easy to come up with. Drag, drop, select. Also, do you want the API providing interpretation of gestures, or do we leave that to the lua people (lua people: a rare breed of humans that spend time creating scripts instead of flying).

I would suggest a gradual approach.

First:

Event: touch
Result: event={action='touch',posStart={x=,y=},posEnd={x=,y=},start=,stop=}

Event: drag
Result: ??

jfrickmann, 2021-05-17:
lshems, take a look at the file /radio/src/touch.h - it defines the events that the system is already using, and it is similar to what you propose.

jfrickmann, 2021-05-17:
@raphaelcoeffic are widgets mandatory, or could we replace them with telemetry screens? That would simplify things alot, and we could provide a telemetry script that could load and run widgets later.

lshems, 2021-05-17:
Yes please

lshems, 2021-05-17:
With key events and autostart. No pagination.just one screen in the main navigation flow. Defaulted to 'auto.lua'.

I think Jesper and I can come up with a LUA alternative for widgets, and/or a standardised way of navigating through

jfrickmann, 2021-07-31:
Can we close this one now?

pfeerick, 2021-07-31:
Yup, since it's now a thing ;)

closed 2021-07-31 by pfeerick