EdgeTX/edgetx #28: pilot-lon, pilot-lan as table keys is very confusing on GPS sensor return

lshems, 2021-05-16:
https://doc.open-tx.org/opentx-2-3-lua-reference-guide/part_iii_-_opentx_lua_api_reference/general-functions-less-than-greater-than-luadoc-begin-general/getvalue

By having a key that contains a minus sign, 'pilot-lon' and 'pilot-lan', referencing the valus is really cumbersome:

getValue("GPS")["pilot-lan"] needs to be used now because getValue("GPS").pilot-lan will try to substract value for variable 'lan' from a value for key 'pilot' . 

Why not change the labels to 'pilot_lon' and 'pilot_lan' and simply use getValue("GPS").pilot_lan ???

raphaelcoeffic, 2021-05-16:
Right, that sounds like an odd thing to do. I just hope we‘re not breaking someone else‘s code.

lshems, 2021-05-16:
made a typo:  pilot_lat it should be of course.

In order not to break, just add the two extra useful labels.

;)

lshems, 2025-11-06:
Should this not be closed?