EdgeTX/edgetx #32: Support normal and expert mode

lshems, 2021-05-17:
I would really be thrilled if the following option would become available from the is GUI.

Normal and expert mode.

What in normal mode?

> Simplified radio setup for only things that you would want to change regularly, like RTC, owner name, mode and calibration.  
> Only nput's labelled as normal, with limited options, like source, weight, expo
> Only mixes labeled as normal, with limited options, like add/multiplying,/replace, source, weight, expo and switches
> No logical switches (but they are named and can be used as sources
> Special functions limited to the normal labelled ones, with a limited set of options, mainly voice commands 
> Only the available outputs as in mike's proposal.
> For all values on those limited parameters, allow every source. Allow for proper 6 or 8 char naming of all sources, if necessary by an external lookup table.

Now in the expert mode, all becomes available without limitations.

In this way, you could create templates with expert  mixer blocks created in expert mode. They can be labeled for use in normal mode. Now I can just take normal mode, assign expert mixes to the outputs that where defined and labelled in the expert mode.(outputs redesign from Mike)

Or Expose the main expert mixes by having them as source on 4 normal mixes for the available 4 outputs.

Want to add stuff to these expert mixes? Just select the sources on four normal muxes to be ailerons, rudder, elevator and flaps expert mixes.
Add rudder input or the rudder expert mix on a second line as source to add rudder to aileron mix in NORMAL mode.
Select the output to be sourced by your new normal mode mixed channel.

GVARS need to be able to be managed much easier. A list with named GVARS showing the values of the active flight mode to be edited. Also only the normal labelled ones, that can be used as adjusters for the params of the expert mixes.

So basically:

1. in normal mode, limit the options in the GUI and the objects defined as normal, but with the expert objects referencable
2. Allow all in expert mode.
3. Support readable naming on all objects.
4. Have fun and just fly

raphaelcoeffic, 2021-05-17:
That sounds like a very good idea! We need to investigate what it means in terms of TODOs, but I like the idea.

lshems, 2025-11-06:
Relevant again??