EdgeTX/edgetx #33: Return autedetect of basic sensors

lshems, 2021-05-17:
Please create a list of 'normal' sensors in a fixed order as we had on 2.1 on fixed positions to be populated on model select.

Whenever more than one sensor is found with the same name, define all as expert sensors, and pop up a warning that expert sensors are used in the model.

This avoids the dangerous situation of sensors being mixed up at start or after rediscovery, after changing hardware sensor configuration.

raphaelcoeffic, 2021-05-17:
I heard this from many different people. The first step would be to generate a consistent ID that would always be the same. Then we can think about auto-detecting these sensors.

lshems, 2021-05-17:
The name passed is enough.

lshems, 2021-05-17:
The sensor manufacturers than will have to sort it out themselves and stay unique, or user annoyance. And duplicate ID cannot be handled anyway.