EdgeTX/edgetx #34: Introduce a 'normal' empty page to allow for telemetry scripts.

lshems, 2021-05-17:
As a first move to killing widgets, create an option to make any screen you define on a colour radio to be a script screen, with only the script name as option.

All key and touch events should be available, even page, menu and system buttons.

Create a proper 'change event' function that is executed from the script BEFORE the main GUI uses the event.

It's up to the scripters to make sure they don't lock up the GUI.

COME ON, a little confidence in the 'lua people'.

(Just don't allow an autostart on that screen, or better, have an escape sequence to boot into the screen setup without running the scripts/widgets first by having menu pushed on startup)

raphaelcoeffic, 2021-05-17:
@lshems just before we start getting into details and designs. Have you tried the new widgets as implemented in 2.4? They offer a fullscreen mode which basically corresponds to many of your wishes. It does require a tiny of user interaction, but I think it comes very close. Basically, when double tipping on any widget, you get a pop-up menu that offer 2 choices: “fullscreen” and “widget settings”. When fullscreen is activated, the widget behaves exactly like the telemetry screens, which is that you get all inputs redirected your way. The only thing ETX keeps for itself, is the [RTN long press], which allows the user to exit the fullscreen mode.

lshems, 2021-05-17:
Well, that sounds nearly like it.

Are there 2.4 nightlies yet?

raphaelcoeffic, 2021-05-17:
> Are there 2.4 nightlies yet?

Sort of. If you have a spare radio with color LCD, you could try it out. Just go to the "actions" tab of our repo page, and you should find binaries.

[image]

[image]

lshems, 2021-05-17:
I'll give that a try 😜

JimB40, 2021-05-26:
Have tried it. Its okay, However it's little bit confusing if I'm in widget full-screen mode or this special full-screen mode.

Still think that @lshems idea is has better UX

Scenario is that we use EdgeTx to configure, once airborne or preparing we use Lua Apps. Some LUA apps even provide OpenTX/EdgeTX configuration so basically less experienced users may end using this app and power switch ;)

1.
Imho this script page should go into full screen widget mode with key sensing without clicking. Any settings in mature LUA apps are done within scripts so no need for widget settings to reconfigure
2.
UI question is how to allow user to switch between views still allowing PAGE buttons to be used by LUA rather then EdgeTX.
My first thought was that swipe left/right is for Edge while buttons are for LUA
3. Even now script in special full-screen mode behaves like normal script it still needs to write additional *.lua telemetry style and describe for user how to install etc.
Simple dropdown list of TOOLS scripts would be much better or even list of SD structure to choose *.lua file (some LUA devs put all files in root subfolder) although this may be much more time consuming to code.

raphaelcoeffic, 2021-05-26:
@JimB40 there still needs to be some support button-only radios ;-)

> Scenario is that we use EdgeTx to configure, once airborne or preparing we use Lua Apps

Fully agreed here, that is somehow the main scenario, at least for most people using LUA. However, they might want to use more than one ;-) I do agree however that it would be better to have a very simple mechanism for full-screen mode.

Basically, we could get rid of the "Widget Settings" at this point (whereby I like it as it offers quick config of regular screens, for regular widgets), and clicking would take you to full-screen, basically removing one step.

So either:
- double tap, you are in full screen.
- or use the roller to select the widget, and click to make it full screen.

And we still keep **[RTN long]** to exit that mode.

JimB40, 2021-05-26:
[RTN long] 100% agree. Not sure if msg shuld be displayed after some time will be annoying.

Still separate full-screen-non-telemetry-script main view would be very convinent.
1. Going to this view it just runs selected script.
2. [RTN long] dimes screen and passes control To EdgeTX
3. now user can [PAGE NEXT/PREV] or swipe left/right to change view or [ENTER], touch or double touch to go back to script.
Or use [SYS],[MDL] or [TELE] to invoke EdgeTX menus still having this view with most used tool in 'background'

So you have let's say one view with iNav, next with Agent and third with traditional widgets layout you have chosen.
/TOOLS with [SYS] shortcut then will be used for rarely used other scripts. 

PS. Dimming on BW radios is not possible on GrayScale (X9D) it is

lshems, 2021-05-26:
Don't forget to disable those stupid popups for resetting stuff. They ruin a fluids experience.

closed 2021-11-11 by JimB40