EdgeTX/edgetx #37: Fix the trim problem that has always existed in OpenTX

Miami-Mike, 2021-05-17:
Somewhere in the OpenTX source code for trims you'll find a constant of 250 that was almost certainly meant to be 256. The result is that the trim limit, which was presumably intended to be 25%, is instead 24.4140625%.

For details please see https://github.com/opentx/opentx/issues/4489 on the GitHub OpenTX forum, and
https://www.rcgroups.com/forums/showthread.php?p=47139791&postcount=41 on RCGroups.

lshems, 2021-05-17:
And fix the bug on the flightmode trim settings API in one go? Or new issue here? Or wait for opentx?

What's the policy?

raphaelcoeffic, 2021-05-17:
> And fix the bug on the flightmode trim settings API in one go? Or new issue here? Or wait for opentx?
> What's the policy?

No "issue highjacking" is the policy ;-)

Miami-Mike, 2021-05-17:
Yeah Guido, don't hijack my issue! ;-)

You should post about that "flightmode trim settings" bug in a new thread and explain what you mean. I wasn't even aware of it myself.

Mike

lshems, 2021-05-17:
Sure. I'll link instead. This seems an easy fix :). 

Sorry...

jfrickmann, 2021-11-20:
#1093

closed 2021-12-07 by pfeerick