EdgeTX/edgetx #38: LUA widget visual glitch / flicker

raphaelcoeffic, 2021-05-17:
Test report by @rotorman:
Running (my MavSDK enhanced) Yaapu LUA screen (https://www.rcgroups.com/forums/showthread.php?3840599-Blog-6-Modified-Yaapu-telemetry-widget-to-support-MavSDK-native-MAVLink-on-OpenTX), I see elements sometimes flicker to foreground that should be on background behind other elements drawn on top - likely an issue with graphics library.

rotorman, 2021-05-17:
https://www.youtube.com/watch?v=UESoQ_J8zmU

rotorman, 2021-05-17:
The strips seem to be only with @olliw42 fork. I compiled today's vanilla ETX and flashed it to my TX16S. With it I see no white stripes as in the video above. So flashed back to @olliw42 fork (with VCP support, thus latest) as I see white stripes again.

raphaelcoeffic, 2021-05-18:
I don’t think this has anything to do with the MAVlink PR. Seems like something not working properly in the double-buffering mechanim.

PR #64 (2021-05-23): Switch the framebuffer address rather than using 2 LTDC layers
This is done according to this article:
- http://www.lucadavidian.com/2017/10/02/stm32-using-the-ltdc-display-controller/

~~Potentially fixes #38 (to be tested)~~.
closed 2021-05-23 by raphaelcoeffic