EdgeTX/edgetx #39: Unable to force shutdown when receiver still connected

raphaelcoeffic, 2021-05-17:
Test report by @rotorman:
When telemetry link is active (RSSI bar shows signal), cannot nicely shut down the radio. No dialog to confirm shut-down, or possibility to exit the Shutdown screen. Deadlock, need to hard-cut power/pull battery.

rotorman, 2021-05-17:
Tested radio was RM TX16S.

pfeerick, 2021-05-18:
Reproducible for me, also RM TX16S, I get the voice-over for 'receiver still connected', the shutdown squares, and then it locks up. No blind stabbing in the dark of `RTN` or `ENTER` just in case the screen isn't shown would unstick it.

raphaelcoeffic, 2021-05-18:
No problem, I‘ll take care! Probably something with the UI loop not handled properly in the pop-up screen.

PR #71 (2021-05-24): Fix alarm popups
Alarm pop-up were broken and would cause a NULL pointer access.

Resolves #39.
Resolves #63.
closed 2021-05-24 by raphaelcoeffic