EdgeTX/edgetx #42: LUA - model.setFlightMode() problem with trimValue #8373 on OpenTX

lshems, 2021-05-17:
So we need to have some guidance here.

So what do we do with OpenTX bugs, probably not solved in OPenTX, like the above?

I would rather shoot on a moving target, than trying to pull a dead horse. But its up to you guys.

lshems, 2021-05-17:
https://github.com/opentx/opentx/issues/8373

pfeerick, 2021-05-18:
Oh, lovely... the can increase but can't decrease trims lua bug... nice feature, that one! 😆 🤮

lshems, 2021-05-21:
Ok, two issues: 

One: if you provide only one table element, it will not update the values, as the current implementation runs over the switches 0-4 of the input provided, so if you ommit input 1, it will not even start updating the values.

Two: it is not correctly interpreting the new value, probably due to a type conversion error (integer to long, ... ???) this results in working only for sitive values, never for negative.

lshems, 2021-05-21:
I will provide a script example this weekend, with some pseudo code on what it does now, and some pseudo code on how it should work in my opinion.

gagarinlg, 2022-09-17:
> I will provide a script example this weekend, with some pseudo code on what it does now, and some pseudo code on how it should work in my opinion.

Is this still an issue?

gagarinlg, 2023-02-25:
Closed, because of missing feedback

closed 2023-02-25 by gagarinlg

lshems, 2023-02-26:
I'm not sure it should be closed. The info is given in the opentx link.

I'll see if can make the script to show the error.