EdgeTX/edgetx #47: How best to handle MULTIPLY on the first mix line?

Miami-Mike, 2021-05-18:
Back in February of 2017 I brought up the fact that in OpenTX 2.1.9, the first line of a mix could be set to MULTIPLY, which would completely disable that line due to the lack of anything above to multiply it by, and there would be no visible cause for the resulting dead mix line because the multiplex setting of the first line in a mix wasn't displayed.

I argued for automatically treating that first mix line as a starting value and not even providing a multiplex option for it, since ADD and REPLACE **should** have identical behavior* and MULTIPLY is pointless, but the developers chose instead to keep the MULTIPLY setting but display a "MULT!" warning. (When you set a first mix line to MULTIPLY and see "MULT!" that's the result of one of my contributions.)

Is the developer's choice the ideal solution? I won't argue one way or the other at this point but it's something you guys might want to consider.

Footnote: ADD and REPLACE don't have identical behavior in some cases where the Slow function is used, but that's a quirk in the Slow function that I intend to address separately.

lshems, 2021-05-18:
Add and replace are operators that require TWO arguments, just like multiplication.

It's just that the minus operator symbol can also be used to indicate a number is negative. Just like the plus operator symbol can be used to indicate a number is positive.

That doesn't make the operators to work without two arguments however. 
Try nil * 5, or nil + 5. 

So by allowing ADD and REPLACE to work at the first position, one has to take as an ASSUMPTION that the first argument would be zero for the ADD, and not relevant but existing for the REPLACE. Then it is obvious to also take a sensible assumption for the first argument for the multiply as first operator in a series.

I vote for 1,one, Uno.

Only issue is see is that when the line is shifted, you need to maintain the original operator and use the preceding line as first argument for the multiplication. This could surprise people, if you would not show the multiply on the first line.

raphaelcoeffic, 2021-05-18:
I would vote for the simplest solution of all: let's ignore the operator (if existing) for the first line. I would have expected this behaviour, in fact, but never tried / tested such thing (I always make perfect mixer programming, right? LoL).

On the display, I would just display the line as-is, no matter where it is placed. This seems to me the most consistent thing to do.

pfeerick, 2021-05-19:
Indeed... I think multi-line operators that depend on a prior line should be disabled on the first line... maybe with it's own error/warning something (similar to the "MULTI!" error/warning mentioned) ... and display the rest as is... then it is shown consistently, unless it is put in that 'special' slot.

PR #5465 (2024-08-26): fix(radio): multiplex setting is not valid for 1st line in a mix
Fixes #47

For the first line in any mix:
- don't use the Multiplex setting in the mixer.
- don't display the Multiplex setting when editing the mix line (radio & Companion).
- remove 'MULT!' warning display (B&W radios and Companion).
closed 2024-08-26 by pfeerick