EdgeTX/edgetx #48: Enable touch control of rc channels

jmil-dev, 2021-05-18:
With the enabling of the touch screen it would be great to allow touch inputs to be processed in a way that allows them to be used as if they were physical switches.

An example may be of an on screen "button" that enables pre-arm of a quad rather than using a physical switch.

I can think of a few options - make a logical switch equal an on screen button, or use Lua to create a widget that would do the same thing

pfeerick, 2021-05-19:
> use Lua to create a widget that would do the same thing

I think that is the plan? Touch screen API => touch input for widgets => widgets can control. I think (hope!) as the API becomes available and people start to use it that we see exactly this 'hybrid' input method... where you can use the touch screen in addition to the physical inputs.

jmil-dev, 2021-05-19:
Perhaps a proof of concept Lua widget would enable you to pick which logical switches were displayed and then let you toggle them.

JimB40, 2021-05-26:
There is very strong reason why LUA in OpenTX cannot change outputs and it is safety. Script or widget can be killed for different reasons by firmware. So changing let's say throttle channel value or arming channel value by something not 100% safe was excluded.
Just to consider.

raphaelcoeffic, 2021-05-26:
> There is very strong reason why LUA in OpenTX cannot change outputs and it is safety. Script or widget can be killed for different reasons by firmware. So changing let's say throttle channel value or arming channel value by something not 100% safe was excluded.

I think we are talking about "secondary functions", like guns, water pump, lights, etc... mostly on boats, which have lots of these gimmicks. The rest should be documented properly: 
```
Do not use LUA for anything essential like motor or basic model functions
```