EdgeTX/edgetx #51: Return button keeps change

olliw42, 2021-05-19:
observed on a Jumper T16:

when doing changes to settings and one is returning by pressing the RTN button, then the changes are accepted nevertheless! They should be accepted only upon a press of return.

This is pretty annoying since it is essentially impossible to jump out from wrong changes.

raphaelcoeffic, 2021-05-19:
I think this is a generic issue on almost all fields: there is no "cancel". I'd have to look into it, but it seems it's going to be a pain to add (lots of things to check). As far as I've seen, the fields are edited directly (changed directly in the model / radio settings). It would require modding libopenui to support "copy-on-edit", and "write-back-on-accept", which it does not at all right now.

Let's tackle this a bit later. But you're right, this is often annoying.

pfeerick, 2021-05-19:
I raised this with OpenTX devs, as this is, and has always been standard behaviour there, and they said... no. I agree that it is often annoying, as you also can't change things live, and then hit return to revert... but I also get it is a bugger of a feature to implement across all the different TX types and interfaces.

olliw42, 2021-05-19:
it is NOT the behavior of otx2,3 !

I frankly would not agree with postponing this, IMHO it is super annoying and just soo standard for any GUI type of thing that it is IMHO just clear what the responses of all users would be concerning if etx2.4 is worth it ...

IMHO this should be a blocker for any release

my 2 cents

raphaelcoeffic, 2021-05-20:
@olliw42 ok, I’ll re-check.

pfeerick, 2021-05-20:
I suggest you revisit OTX 2.3, as you will find it *is* the behaviour, and has always been that behavior of OTX since I started using it in the 1.x days with my Turnigy 9X!! It's not new! If you make any changes to a setting whatsover, pressing return will save it, the same as pressing enter will. Furthermore, you can see the devs say *why* they refused to change from the current behaviour, to what you are suggesting, here. https://github.com/opentx/opentx/issues/8369

olliw42, 2021-05-20:
> I suggest you revisit OTX 2.3, as you will find it _is_ the behaviour

that's indeed correct, my fault 

but it doesn't change my opinion how it should be ... 

the arguments provided by some devs in the ref you gave, like "that's how every R/C radio out there works" or "it would cause massive confusion with users especially since everything IT is moving to a similar "instant save" model nowadays", are - IMHO - just nonsense and do not appear to be generally true

pfeerick, 2021-05-20:
Indeed... that's why I gave up on the issue... 

To be fair, I can see both arguments being true... as it would mean a change in a behavior that those of us who have been using it for for a long time could lead to confusion or even a plane crashing... because as it was pointed out, sometimes it is necessary to make those changes in flight for tuning, and having return now suddenly revert a setting could be quite disastrous since until now it was automatic/instinctive that that all that would happen is the field being edited left, not the edit reverted. 

Which is why I then suggested it should be an option, but the question naturally then was 'who would implement the feature' ... I would have preferred they get 2.4 out the door before working on 'would like' features. Given the amount of work required, and that it would almost certainly be a blocker for re-integration into opentx post 2.4, I do not think this should be considered or implemented yet, and there are much more important issues to be worked on given the proposed release date is just over a month away. 

Unless I am mistaken the main goal goal of this first release is to basically do what was promised - OpenTX 2.4 with colour touch screen support. After that, enhancements like this can/should then be assessed, discussed, and assigned to milestones to clearly indicate the direction things are going.

JimB40, 2021-11-26:
To be closed imho. Or moved to dicuss.

pfeerick, 2021-11-26:
Closed, no, not a chance - this is why it was raised here, as that is what happened over at OTX. Moved, most likely, as will quite a few issues I think ;)

MervDale, 2021-11-26:
I'll bite , doesn't having to use the Return key negate somewhat the functionality of a Touchscreen,  one example being  the Enable Box, the mere fact that touching this box selects it but if you then are required to press the Return Key you will deselect it,?

pfeerick, 2021-11-26:
Please don't... there's only like three transmitters out of 20 that have touch screen... so that basically throws that argument out the window :-P You might want to read up on this issue a bit - it's in relation to the historical behaviour of RTN, not ENT, on saving of data.

closed 2021-11-26 by pfeerick