EdgeTX/edgetx #52: MDL Button long press doesn't go back

olliw42, 2021-05-19:
observed on a Jumper T16:

previously it was possible to navigate forward and backwards by pressing e.g. the MDL button short or long. 

The pressing long and go backwards features is not possible anymore now.

pfeerick, 2021-05-19:
Yes, it was the page button... long press for back, short press for forward. I'm not sure about MDL... I think it would talk you to the channel monitor?

olliw42, 2021-05-19:
ah, yo .. LOL
but you in principle could have that behavior on ALL buttons, the code did handle all equally in otx2.3 ... so it's only a matter of wether it is used in the GUI or not

pfeerick, 2021-05-20:
I have not idea what you're trying to say there... would you please try explaining that again. 

I believe you're trying to say that behaviour could be applied to any button, which is true, but the question would be *why*? Since on the T16 your keys are SYS, RTN, MDL, PAGE, TELE and enter (the roller button)... it really doesn't make sense for any other key to be 'page back'. In this respect, it is behaving just like the X9D(+). The TX16S is either the only, or one of a very few 'modern' TXs that has dedicated page forward and back buttons - and I for one wouldn't actually mind if holding the page forward key triggered page back *as well as the back key* ... since it's the odd transmitter of the bunch. It's actually rather funny in a way... since the the older 9X and 9XR-PRO only had six buttons (less if you consider the roller a three button interface) and the two horizontal buttons on the four way pad were the page forward and back buttons.

olliw42, 2021-05-20:
1) architecture wise it is ugly design and bad coding practice to implement a functionality just for one thing even though it applies to all members of the class of the thing
2) even if now one may only see it being useful or used for one thing, it's just a lack of imagination to consider that it  might be useful later on
3) it is simply not true it is used - in otx2.3 - only for page
4) it is not about forward and backwards but short and long

and just to have said it, I'm using it for sure for my lua script for the MAVLink for OpenTx ;)

pfeerick, 2021-05-20:
Ah, yes, I'm with you now... it was just a _bit_ confusing because your issue started about about long press not going back, when in fact it's more about not having the facility to use short and long presses for the interface generally. 

Yes, it is certainly ugly... and is yet another thing that makes me hope ETX makes it to v3, consisting of a complete re-write / overhaul of the input and menu/navigation system behind the scenes, using more modern C++ object oriented techniques right from the start... meaning you would have button objects with short, long, double press etc, events out of the box essentially.

olliw42, 2021-05-20:
btw, the missing long gives etx2.4 a total strange and unnatural behavior which I really don't like in that you can press a button and nothing happens ... until you release it ... note e.g how differently the sys button works for etx2.4 and otx2.3 when you just keep it pressed ... in otx2.3 it is obviously connected to long ... (similar for other buttons/situations)

closed 2021-05-20 by olliw42

pfeerick, 2021-05-20:
Nice. Since it looks like Raphael actually changed it to break from the prior long press setting, maybe he preferred short presses, or to have no delay to speed up testing. 

Also, better in future to leave the issue open, as it's not really resolved until PR is merged. Make use of the github keywords automation of 'Fixes #issuenum' / 'Closes #issuenum' / Resolves #issuenum' to have it auto-close on merge ;)

olliw42, 2021-05-20:
> maybe he preferred short presses

if so it wouldn't have made sense to have long for KEY_TELEM but not the others

also, the correct way would then have been to check for break and long, not only break

(which reminds me that the PR might need to be updated, since the breaks should probably be break&long...)