EdgeTX/edgetx #53: In OpenTX you can't turn the backlight on to read a model note because it won't turn on until you dismiss the note.

Miami-Mike, 2021-05-19:
[I posted this on the OpenTX GitHub forum](https://github.com/opentx/opentx/issues/6507) in June of 2019 and there was no response. It was eventually marked stale, and the problem still exists in OpenTX 2.3.11.

If you have the backlight set up to turn on and off with a special function, and you have a model note set up to appear when you turn the radio on or select a model, you can't turn the backlight on until you dismiss the model note. It's a classic [Catch-22!](https://en.wikipedia.org/wiki/Catch-22#Concept)

raphaelcoeffic, 2021-05-19:
Indeed! I don't use that function, but if I would, that would really annoy me!

So basically, the model notes are displayed, before the special functions are active, right?

Miami-Mike, 2021-05-19:
Right, that appears to be what's going on.

pfeerick, 2021-11-28:
It looks like the main part of this issue - that you can't turn the backlight on if assigned to a SF has been resolved, as I can turn the backlight on and off at will whilst in the model checklist. It looks like _forcing_ the backlight on is going to be a pain to implement though.

Eldenroot, 2023-01-11:
@gagarinlg - this mentioned PR is still a draft, not merged... I am waiting for it! ;)

gagarinlg, 2023-01-11:
> @gagarinlg - this mentioned PR is still a draft, not merged... I am waiting for it! ;)
sorry

commit db410eb2 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit 8798b51e (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit ba0fb945 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit ab3cb457 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53