EdgeTX/edgetx #54: Add an indicator icon for haptic vibration to the radio screens.

Miami-Mike, 2021-05-19:
This is another issue that [I raised in the OpenTX GitHub forum](https://github.com/opentx/opentx/issues/6483) which was ignored and eventually marked stale.

When you set up haptic vibrations in Companion, either on the Radio settings Setup tab, or with a special function, or through a Lua script, there's no indication in Companion simulator when haptic vibrations are active.

A little animated icon on the radio screen would work. It would not only indicate haptic activity as it happens in Companion simulator, it would also indicate on the actual radio screen that the system is at least trying to create vibration. In that case, when vibration doesn't actually happen the user would know that the radio has a hardware problem.

olliw42, 2021-05-19:
I must frankly say I would not like it. The space in the top bar is already quite cramped, and I would use it rather for really important stuff and not for just convenience indicators.

I guess what would probably more suitable would be some lua script which one could put into the user-configurable slots.

just my 2 cents :)

Miami-Mike, 2021-05-19:
Which radio is cramped for space in the top bar? I looked at few different radio screens in Companion simulator and all of the ones I looked at have room. In fact, it seems like all but the 128x64 screens have quite a bit of room.

pfeerick, 2021-05-20:
It's probably not that hard to do either... it could either be an indicator, animation of an existing indicator like the speaker volume icon (i.e. it could be swapped with a vibrate icon), or other basic UI indicator like the top bar flash with the haptic vibration, or a outline box around the whole screen. 

In other words, a visual indicator to go along with the haptic feedback.

pfeerick, 2021-05-20:
> Which radio is cramped for space in the top bar? I looked at few different radio screens in Companion simulator and all of the ones I looked at have room. In fact, it seems like all but the 128x64 screens have quite a bit of room.

On the colour screen radios, the top bar has the logo, four slots for on-screen widgets that are visible regardless of which telemetry page you are on, and then things like the RF signal, battery bar, volume, and date/time... This will soon also have a GPS widget if the radio has an internal GPS enabled. So adding a specific indicator 'slot' could chew into the space for the four widget bars.

Then again, if the icons that were shown were customizable... that would be a different matter entirely... i.e. when you go to the panels screen, you can turn all manner of things on or off... trims, sliders, flight mode... even the top bar... so why not also the icons shown? There's a whole empty space on the UI page that could have checkboxes for that ;)

olliw42, 2021-05-20:
> Then again, if the icons that were shown were customizable... 

there are these slots for widgets, so all it would need is a widget for that thing ... 
that's the idea of them, right, to make it customizable according to what a user wants, and not by xillions of flags to customize

tekikai, 2022-07-23:
I think what @Miami-Mike meant was a visual indicator in the Companion App or rather Simulator when haptic feedback is active. 
I don't think that this needs to be on the actual simulated screen of the radio but just some sort of indicator light on the radio simulator interface.

I am currently trying to write my own haptic patterns and it would be great if one could at least see (or add a buzzing sound sample to the simulator) or even hear the buzz from the playHaptic() function. Because as it is right now I basically have to upload it to the radio every time I would like to try and hear / feel the pattern.

pfeerick, 2026-09-28:
There is now haptic "feedback" thanks to #7322

closed 2026-09-28 by pfeerick