EdgeTX/edgetx #60: Tools incorrectly populated

olliw42, 2021-05-20:
this can happen every once in a while
seems to be more likely when on is pressing various keys fast

[image]

rotorman, 2021-05-22:
Happens on my TX16S similarly. The fields to the right of lua are filled with random stuff it seems, and not with the names of the LUA scripts from the SD card TOOLS subfolder.

rotorman, 2021-05-22:
Here is a video showing this. I was regularly able to induce this on TX16S, by hitting:
1) System key
2) Page > key
3) Page < key

https://www.youtube.com/watch?v=RVlaJLJoL8A

raphaelcoeffic, 2021-05-22:
Here I think it's even worst, as my SD card is full of MacOS garbage (`._whatever` files....), which seems to confuse the file scanning. I'll give it a shot.

olliw42, 2021-05-22:
that's another thing, that I now constantly have to repair the SD card ... (even in normal operation)

raphaelcoeffic, 2021-05-22:
> that's another thing, that I now constantly have to repair the SD card 

That's not normal. Never had that. Maybe the card is starting to fail?

closed 2021-05-29 by raphaelcoeffic

raphaelcoeffic, 2021-05-29:
Please note that the script are not yet working all correctly, as there is still some issue with the file loading. The "tools" screen is now correctly populated however.

commit b79c459b (2021-05-29): Fix fetching LUA tool name

Resolves #60