EdgeTX/edgetx #61: In OpenTX, when reading the value of an input the effect of trim is omitted.

Miami-Mike, 2021-05-21:
Set up a "Numbers" telemetry screen that monitors a stick input and you'll see that the effect of trim is omitted. There's no way to directly monitor the value that an input sends to a mix while it has a non-zero trim setting. 

The lack also exists in Lua: [Input1 (lua script).txt](https://github.com/EdgeTX/edgetx/files/6523916/Input1.lua.script.txt)

I don't know if the OpenTX developers would be interested in fixing this because it's been this way at least since 2.1.9 and has probably always worked like this, but it would be nice to see it done right in EdgeTX.

lshems, 2021-05-21:
A stick input is different from an input in the inputs screen.
If you want to show a stick input with trims, set up an input and have the trim added option set.

A stick input value should be 'raw' in lua, and trims should be 'raw' as well, and added in lua itself when needed.

I don't think you can say the trim is 'omitted'. stick input is simply defined as that: stick input. as trim input is trim input.

Just my point of view.

Miami-Mike, 2021-05-22:
Sorry, apparently I wasn't clear because I couldn't make sense of your response. I'm talking about an input on the Inputs screen for throttle, rudder, elevator, or aileron, with "Include Trim" set to the default of "Yes".

You can modify the value of an input with weight, offset, one of the Curve options, etc. and the resulting value, which is the value fed to a mix as its source, will show correctly on a "Numbers" telemetry screen and be readable with Lua's getValue() function.

But if you move the trim away from zero, the value displayed on the telemetry screen or read by Lua won't change and will no longer be the correct value that the input is producing and which will be fed to a mix as its source. In fact, there is no way to display the correct value that the input is producing and which will be fed to a mix as its source.

lshems, 2021-05-22:
This should be fixed in immediately!

Lol. Indeed, terminology got mixed up.

Stick input is raw data.
Input line with stick input as source should reproduce what goes into the mixer wheen displayed or used in lua getValue.

It are these kind of errors that are one of the reasons of scripting being so horrible.

Thanks for clearing it out.

olliw42, 2021-05-22:
I would strongly argue that trim should not be applied to any input but only output, if a shift is needed permanently one should add an offset (unless some specific setting says that the trim value should be mixed to the input)
so the correction to what you consider wrong behavior I would consider a bug ;)

I frankly think that concerning all the mix etc things you guys are on a dangerous path.
1. OpenTx (and other similar projects) has spend quite some effort to work out their scheme some years ago which implies that the results is probably for good reasons ... it might be correct to call it one of the most matured part
2. in many places there is no clear wrong or right, but different possibilities of what one may expect to be wrong or right, which implies that some users will expect it the one way and others the other way ... ergo, not all user's expectations can be satisfied and ergo a decision needs to be made

I would recommend to be very careful and hesitant with changing zillions of these aspects and in each case claim that it's a bug or an issue ... first because OpenTx mixings are quite developed and mature and proven in the field and second because all too often it is just that someone's individual expectation is not met

the handling of scales is simple, but the handling of offsets (trims) is quite difficult by principle, as much as I can see

lshems, 2021-05-22:
Again misunderstanding.

Nobody wants to alter the mix behaviour.

It's only the available values referenced in telemetry and lua that are NOT reflecting the (correct) input to mix behaviour.

Miami-Mike, 2021-05-22:
"I would strongly argue that trim should not be applied to any input but only output..."
"I frankly think that concerning all the mix etc things you guys are on a dangerous path..."

This thread is about how a certain value is displayed on a "Numbers" telemetry screen and how its value is read by Lua. It has nothing to do with how the operating system actually works.

I won't respond to your comments here, but I'll say that due to the way GitHub is structured, I think your comment belongs in a thread of its own and not mixed in with this one.

pfeerick, 2021-07-03:
I'd also have also pointed out that nothing has been changed in the mixer or inputs settings... this is all as-is from OpenTX. 

It's pretty obvious that if you have the 'Include Trim' option ticked, and it doesn't actually get applied as far as the value that is fed into Lua or the numbers telemetry view that something is broken for that 'Input', as the applied trim setting isn't been passed on. After all, what is the point being able to access the Input values if they're not correct in all cases?