EdgeTX/edgetx #63: Powering down with usb connected -> weird state

olliw42, 2021-05-22:
when powering down with a usb connected, it ends up in a weird state, and the transmitter does not actually switch off

it IMHO either should not allow powering down when the usb is connected, or terminate all properly

raphaelcoeffic, 2021-05-22:
@olliw42 can you please precise your report? Is that happening in the bootloader or the in the main app? Which USB mode did you choose?

rotorman, 2021-05-22:
I can confirm that the non-power off happens also with d3189b38e6862f60075a1cd07bc11a1e946c592e already (just had it still flashed on my TX16S), and even if I choose Joystick (HID) as USB mode.

I believe the weird state (=black screen, with backlight on, power button LED green) is the Emergency Mode screen, and that EM has no content yet in EdgeTX.

olliw42, 2021-05-22:
fair enough
here the steps to reproduce

- power up the radio as normal
- connect a usb
- the popup selector opens, but don't do anything
- press the power button and wait

it might well be that the issue or similar issue might occur with other recipies

I have difficulties to describe the weired final state, it doesn't power down and some blanked screen remains

pfeerick, 2021-05-22:
I can't reproduce that on the TX16S with current https://github.com/EdgeTX/edgetx/commit/61abc46547bf6519ef90429c4a0c88148ac41f9a commit. Neither not selecting and option, or selecting either of the USB options behaves abnormally. Well, any more abnormally than usual - the TX16S will always leave the flight mode led lit after powering off if USB power connected. No power light, no emergency screen, no backlight. 

This sounds somewhat similar to the 'receiver still connected' issue - transmitter was stuck with a blank screen, backlight still on, and you had to pull the battery to turn it off.

rotorman, 2021-05-22:
I just remembered, that my TX16S is "special" what top USB-C goes, as I have added a physical DFU button, where STM32F429BI BOOT0 is not hard-wired connected to VBUS as the RM default ( https://www.rcgroups.com/forums/showthread.php?3845213-Blog-9-RadioMaster-TX16S-DFU-button-modification ). Thus, I might be not so good candidate testing this.

PR #71 (2021-05-24): Fix alarm popups
Alarm pop-up were broken and would cause a NULL pointer access.

Resolves #39.
Resolves #63.
closed 2021-05-24 by raphaelcoeffic