EdgeTX/edgetx #67: Can we get rid of the last functional legacy terminology of MODE and RETA definitions?

lshems, 2021-05-22:
In principle the four sticks are not predestined for Rudder, Elevator, Throttle and Ailerons, as we can use the radio equally well for drones, that don't even have ailerons for example.
The alternative of Yaw, Pitch, Throttle and Roll doesn't work either in all cases, since that doesn't make sense for drones. There is no 'correct' choice possible.

So after some consideration I think the stick inputs should nonetheless be renamed to use the desired motion of a simple plane using these control inputs, rather than naming them after the control surfaces that on a basic plane would generate that motion. We should reserve those for the actual outputs, that would transfer the control inputs to the desired movements of the control surfaces, who would in turn lead to the desired motion.
Mixes would then allow to make corrections, necessary due to the physical limitations/behavior of the craft, to create an as pure as possible transfer of desired motion to actual motion. Think of mixing some yaw into the aileron output, so that roll input would create a pure roll, and no barrel roll.

The above would have impact on several things:

If you create a default model, that is, with the wizard or just without an SDcard using the radio default settings, the mix flow is currently the following:
`stick source -> inputLine with trim ON and names -> mixerLine with trim ON -> outputLine without names`
The ordering will follow the radio setting, so RETA, RAET, whatever, from input to output. It is NOT taking the mode into account.

There are several things I don't like with this. It seems a bad compromise to me of straightforward technical mixer setup from input to output with the legacy remains or appearance of a functional plane setup. 
I would propose to either do a PURE technical setup, or do a correct functional setup.

PURE technical setup, no inputLines needed:
`stick source -> mixerLine with trim ON -> outputLine`
mixerLines ordered based on the stick sources, following the RETA definition in the Radio settings. No names, labels or nothing. Here you can see how the current definitions would actually link the Rud, Ele, Thr and Ail input sources to the output control surfaces Rudder, ELevator, Throttle and Aileron. It was how it was when we had no computer mixers.

Correct functional setup:
I assume the stick sources are renamed hardcoded to Yaw, Pitch, Throt and Roll.
People can still label them as Rud, Ele, Thr and Ail if they want to stick with legacy names, or name them Yaw, Forward, Throttle and LeftRight when flying drones.
That labeling should by the way be by preference be on a MODEL level, not on radio level. I would omit it all together, and stick with only Yaw, Pitch, Throt and Roll. 

So for the RADIO created default model (no wizard) I would use then
stick source -> inputLine with trims -> mixerLine -> outputLine with name
So no naming on the inputLines, and certainly no confusing trim ON on the mixerLines. The inputLine order would follow the MODE, so
1. Yaw, Pitch, Throt, Roll
2. Yaw, Throt, Pitch, Roll
3. Yaw, Pitch, Throt, Yaw
4. Roll, Throt, Picth, Yaw
The mixerLines would follow the RETA definition on the radio setup. The outputLines are named accordingly to prevent confusion. The mixers would do the exact mapping. The mixer for the Elevator would have a weight of -100%, so that all control surfaces would comply with the very important notion of the mixer output being positive for desired positive output surface deflection on corresponding stick input. 

Why is this important?

First of all, for newcomers, This dual use of the same names for inputs and outputs is confusing. Als, using the trim flag is SUPER confusing being present on both inputs and mixers, where they have no role on the mixes in a setup where the source of the mix is an inputline.
Second, if we want to start newcomers to understand the basics, better have it correctly applied from the start. Using the desired behavior as label for the control inputs and the naming resulting outputs after the surfaces you want to control is standard practive in all control system setups.

I will attach some screenshots to clarify.

I know this is not a High Prio request, but I think you must not underestimate the impact it might have to get newcomers to understandthe basics of OpenTX.

lshems, 2021-05-22:
[image]
For me, the labels on the right should be the hardcoded ones
[image]
The input names are not really neccesary in thsi screen, but will improve readability on the mixer screen.
[image]
Here the names of the resulting mixes are given, so that they can be referred to on the outputs page according to the proposal by Mike
[image]
The output names would actually be the references to the mixers, according to Mikes proposal.

raphaelcoeffic, 2021-05-23:
This is indeed pure legacy, and probably just wrong use of terms. At the same time, you're right, not really high prio. If enough people vote for it, then let's do it.