EdgeTX/edgetx #74: EdgeTX Logo briefly visible between multiple warning messages at boot-up

rotorman, 2021-05-24:
The following video shows this at roughly 9 second mark:
https://www.youtube.com/watch?v=TEQGIJDvzvU

During startup, when multiple warning messages are shown (here e.g. throttle and SD card), briefly EdgeTX logo is visible, when the display switches from one warning message to other.

raphaelcoeffic, 2021-05-25:
Strangely, this happens only when the keys are used to discard the warnings. I logged the process on serial debug with both variants.

Here with the keys (showing the problematic behaviour):
```

#############
# with keys #
#############

+00000ms: 3.77 # Layout2P1 refresh: Layout [0, 0, 480, 272]
+00000ms: 3.77 # Layout2P1 refresh: Layout [480, 0, 480, 272]
+00000ms: 4.27 # Layout2P1 refresh: Layout [0, 0, 480, 272]
+00000ms: 4.27 # Layout2P1 refresh: Layout [480, 0, 480, 272]
+00000ms: 4.77 # Layout2P1 refresh: Layout [0, 0, 480, 272]
+00000ms: 4.77 # Layout2P1 refresh: Layout [480, 0, 480, 272]
+00000ms: 5.27 # Layout2P1 refresh: Layout [0, 0, 480, 272]
+00000ms: 5.27 # Layout2P1 refresh: Layout [480, 0, 480, 272]
+00000ms: 5.77 # Layout2P1 refresh: Layout [0, 0, 480, 272]
+00000ms: 5.77 # Layout2P1 refresh: Layout [480, 0, 480, 272]
+00000ms: 5.95 FullScreenDialog [0, 0, 480, 272] received event 0x602
+00000ms: 6.12 FullScreenDialog [0, 0, 480, 272] received event 0x202
+00000ms: 6.12 ViewMain [0, 0, 480, 272] setFocus()
+00000ms: 6.12 FullScreenDialog [0, 0, 480, 272] onFocusLost()
+00000ms: 6.12 [ViewMain] Focus gained
+00000ms: 6.12 Layer removed for FullScreenDialog [0, 0, 480, 272] (1 layers)
+00000ms: 6.12  - MainWindow [0, 0, 480, 272] (focus=ViewMain [0, 0, 480, 272])
+00000ms: 6.12 Delete 0xd02142a8 FullScreenDialog [0, 0, 480, 272]
+00000ms: 6.12 Delete 0xd02165b8 FabButton(1) [396, 188, 68, 68]
+00000ms: 6.12 New layer added for FullScreenDialog [0, 0, 480, 272] (2 layers)
+00000ms: 6.12  - MainWindow [0, 0, 480, 272] (focus=ViewMain [0, 0, 480, 272])
+00000ms: 6.12  - FullScreenDialog [0, 0, 480, 272] (focus=---)
+00000ms: 6.12 FullScreenDialog [FormGroup] [0, 0, 480, 272] setFocus(0)
+00000ms: 6.12 FullScreenDialog [0, 0, 480, 272] setFocus()
+00000ms: 6.12 ViewMain [0, 0, 480, 272] onFocusLost()
+00000ms: 6.13 [ViewMain] Focus lost
+00000ms: 6.13 Refresh full screen
+00000ms: 6.13 MainWindow [0, 0, 480, 272] (skipped)
+00000ms: 6.13   SwitchWarnDialog [0, 0, 480, 272] (*)
+00000ms: 6.13     FabButton(1) [396, 188, 68, 68]
+00000ms: 6.13 ViewMain [0, 0, 480, 272] setFocus()
+00000ms: 6.13 SwitchWarnDialog [0, 0, 480, 272] onFocusLost()
+00000ms: 6.14 [ViewMain] Focus gained
+00000ms: 6.14 Layer removed for SwitchWarnDialog [0, 0, 480, 272] (1 layers)
+00000ms: 6.14  - MainWindow [0, 0, 480, 272] (focus=ViewMain [0, 0, 480, 272])
+00000ms: 6.14 Delete 0xd0217a40 SwitchWarnDialog [0, 0, 480, 272]
+00000ms: 6.14 Delete 0xd0217248 FabButton(1) [396, 188, 68, 68]
+00000ms: 6.15 SD card version mismatch:  2.4V0036, 2.4V0037
+00000ms: 6.15 New layer added for FullScreenDialog [0, 0, 480, 272] (2 layers)
+00000ms: 6.15  - MainWindow [0, 0, 480, 272] (focus=ViewMain [0, 0, 480, 272])
+00000ms: 6.15  - FullScreenDialog [0, 0, 480, 272] (focus=---)
+00000ms: 6.15 FullScreenDialog [FormGroup] [0, 0, 480, 272] setFocus(0)
+00000ms: 6.15 FullScreenDialog [0, 0, 480, 272] setFocus()
+00000ms: 6.15 ViewMain [0, 0, 480, 272] onFocusLost()
+00000ms: 6.16 [ViewMain] Focus lost
+00000ms: 6.16 Refresh full screen
+00000ms: 6.16 MainWindow [0, 0, 480, 272] (skipped)
+00000ms: 6.16   FullScreenDialog [0, 0, 480, 272] (*)
+00000ms: 6.18     FabButton(1) [396, 188, 68, 68]
+00000ms: 6.18 MainWindow::run took 23ms
``` 

And with touch, for comparison (works just fine):
``` 
##############
# with touch #
##############

+00000ms: 3.77 # Layout2P1 refresh: Layout [0, 0, 480, 272]
+00000ms: 3.77 # Layout2P1 refresh: Layout [480, 0, 480, 272]
+00000ms: 4.27 # Layout2P1 refresh: Layout [0, 0, 480, 272]
[... 40 lines cut ...]
+00000ms: 4.56 [ViewMain] Focus lost
+00000ms: 4.56 Refresh full screen
+00000ms: 4.56 MainWindow [0, 0, 480, 272] (skipped)
+00000ms: 4.56   FullScreenDialog [0, 0, 480, 272] (*)
+00000ms: 4.58     FabButton(1) [396, 188, 68, 68]
+00000ms: 4.59 MainWindow::run took 24ms
+00000ms: 4.77 # Layout2P1 refresh: Layout [0, 0, 480, 272]
+00000ms: 4.77 # Layout2P1 refresh: Layout [480, 0, 480, 272]
+00000ms: 5.27 # Layout2P1 refresh: Layout [0, 0, 480, 272]
+00000ms: 5.27 # Layout2P1 refresh: Layout [480, 0, 480, 272]
+00000ms: 5.68 FabButton(1) [396, 188, 68, 68] setFocus()
+00000ms: 5.68 FullScreenDialog [0, 0, 480, 272] onFocusLost()
+00000ms: 5.68 Refresh full screen
+00000ms: 5.68 MainWindow [0, 0, 480, 272] (skipped)
+00000ms: 5.68   FullScreenDialog [0, 0, 480, 272]
+00000ms: 5.70     FabButton(1) [396, 188, 68, 68] (*)
+00000ms: 5.71 MainWindow::run took 25ms
+00000ms: 5.71 ViewMain [0, 0, 480, 272] setFocus()
+00000ms: 5.71 FabButton(1) [396, 188, 68, 68] onFocusLost()
+00000ms: 5.71 [ViewMain] Focus gained
+00000ms: 5.71 Layer removed for FullScreenDialog [0, 0, 480, 272] (1 layers)
+00000ms: 5.71  - MainWindow [0, 0, 480, 272] (focus=ViewMain [0, 0, 480, 272])
+00000ms: 5.71 Delete 0xd02182f8 FullScreenDialog [0, 0, 480, 272]
+00000ms: 5.71 Delete 0xd0218258 FabButton(1) [396, 188, 68, 68]
```

pfeerick, 2021-05-25:
> Strangely, this happens only when the keys are used to discard the warnings.

If I power on with the throttle above minimum, and use touch only, I get a flash of logo between the throttle warning and the sd card warning...

pfeerick, 2021-05-26:
I should be able to try it out in about an hour.

pfeerick, 2021-05-26:
Oh, very interesting... I am already running https://github.com/EdgeTX/edgetx/commit/945fc9510ae451325803b6d2c1ba2ec15b014079, so was surprised when you said it was gone, as it's not for me... and it is... 

tl;dr It appears to only appear between the throttle warning and the SD warnings, doesn't appear if the switches warning is triggered also. And it doesn't make any different if the throttle warning is dismissed by touch, button or moving the stick. 

**Example 1:** 
Switches out of wack, throttle *up*, wrong SD version file = correct the switches and throttle/touch/press enter, press roller or touch to dismiss SD warning, no logo flash between warnings. 

**Example 2:** 
Switches out of wack, throttle *down*, wrong SD version file = correct the switches/touch/press enter to dismiss, press roller or touch to dismiss SD warning, no logo flash between warnings. 

**Example 3:** 
Switches where they belong, throttle *up*, wrong SD version file = correct the throttle/touch/press enter, press roller or touch to dismiss SD warning, logo flashes between warnings.

rotorman, 2021-05-26:
I can confirm what @pfeerick wrote, is exactly what I see on my TX16S as well. The EdgeTX logo flicker is there between the warning screens **only, if the switch warning is not shown** (e.g. between throttle and sd card, but not when switches are also not in their correct position).
I do see the flicker also, when acknowledging with a touch screen!

pfeerick, 2021-05-27:
Further test with later build https://github.com/EdgeTX/edgetx/commit/f58317f74ed3483e0a1d0384801bcc6c3b62933f ... Now I have a full sd pack meaning no version warning... this now presents as a logo, 'throttle warning' screen, momentary flash of logo and finally main screen.

rotorman, 2021-05-27:
Same here with e903a88f - with throttle warning, I get briefly EdgeTX logo flash to foreground before the main screen is shown.

gagarinlg, 2022-09-16:
this does not happen anymore

closed 2022-09-16 by gagarinlg

PCS-Hazak, 2026-01-28:
## Investigation Update: LUA Scripting & GUI Reference

Added third investigation file completing the LUA development toolkit:

| File | Topic |
|------|-------|
| `pcs/investigations/2026-01-27_model-consolidation.md` | Model pack options (13→5) |
| `pcs/investigations/2026-01-27_msp-over-crsf-capabilities.md` | MSP command catalog (60 functions) |
| `pcs/investigations/2026-01-28_lua-scripting-gui-reference.md` | **NEW** - LUA scripting + GUI for RadioMaster Pocket |

Together these three files provide complete context for building MSP-based LUA tools on the Pocket.

gagarinlg, 2026-01-28:
Did you answer to the wrong ticket?