EdgeTX/edgetx #75: Flashing bootloader from EdgeTX fails (at least with RM TX16S)

rotorman, 2021-05-24:
Flashing radio bootloader from running EdgeTX presently fails (at least with RM TX16S and with efdf76a9).

As you can see in the video below, it gives SD card error (video 0:15 mark), although when flashing the same _bootloader.bin_ with STM32CubeProgrammer and then flashing the radio binary (efdf76a9)using this bootloader from the same card, all works. Otherwise also, no errors with this specific microSD card.

https://www.youtube.com/watch?v=GHqrVRQjqs8

As you can also see, after pressing RTN, _Flash successful_ is wrongly output (0:20 mark). When trying to boot into bootloader (trims T1+T4 together, power-on), bootloader does not start, thus flashing was unsuccessful (0:33 mark). Have to pull battery/cut power to shut off.

raphaelcoeffic, 2021-05-24:
Not bad! This needs some serious fixing!

dklueh79, 2021-05-27:
Workaround:
you could flash it easily using the OpenTX 2.3.x Companion and the nightly *bin file for the TX16s. Just choose the the RM profile and flash firmware-file. works.

rotorman, 2021-05-27:
Flashing solved in PR #103.

closed 2021-05-27 by rotorman

PCS-Hazak, 2026-01-28:
## LUA Scripting & GUI Reference Complete

**Local file:** `pcs/investigations/2026-01-28_lua-scripting-gui-reference.md`
**Companion to:** `pcs/investigations/2026-01-27_msp-over-crsf-capabilities.md`

### What's Covered

Complete reference for building LUA GUI tools on RadioMaster Pocket (128x64 B&W):

- **Script types:** Tool scripts (exclusive LCD, MSP tools) vs Telemetry scripts (background + foreground)
- **LCD API:** All drawing functions, font sizes, flags, coordinate space
- **Event handling:** Button mapping for Pocket (SYS, MDL, RTN, PAGE, TELE, wheel)
- **CRSF/MSP transport:** `crossfireTelemetryPush(0x7C, frame)` with helper functions and rate limiting
- **Input reading:** Switches, sticks, pots, logical switches, telemetry sensors
- **Model API:** GVARs, timers, logical switches, module detection
- **Audio:** WAV playback, TTS, tones
- **GUI patterns:** Menu with cursor, confirmation dialog, progress bar, key-value display, scrollable list, multi-page nav, state machine UI
- **Complete example:** Working "FLEX MSP Tool" script skeleton (menu → confirm → send → result)

### Key Constraints

| Constraint | Detail |
|-----------|--------|
| No widgets | B&W LCD, not color — Tool/Telemetry scripts only |
| No file I/O | Can list dirs but NOT read/write file contents |
| No callbacks | Polling model via `run(event)` only |
| Tool name limit | 16 chars in Tools menu |
| MSP rate | Space sends 50-100ms apart |

### Cross-References

- MSP command catalog: `pcs/investigations/2026-01-27_msp-over-crsf-capabilities.md` (see #75 comment from 2026-01-27)
- Model consolidation: `pcs/investigations/2026-01-27_model-consolidation.md`
- Existing scripts analyzed: `elrsV3.lua`, `PCS_INAV.lua`, `iNav.lua`

PR #103 (2021-05-29): Allow flashing a bootloader which does not end on 1KB boundary or is …
…shorter than the maximum size

Resolves #75
commit 954758c7 (2021-05-29): Allow flashing a bootloader which does not end on 1KB boundary or is shorter than the maximum size

Resolves #75