EdgeTX/edgetx #76: TX16S, adding new view with non-default layout, top 2 configuration elements not reachable with roller anymore

rotorman, 2021-05-24:
Tested on TX16S with 88e62e1f.

When adding a new view with non-standard layout, after changing the layout type, top two elements (Layout type itself and Button "Setip widgets") are not reachable anymore with roller.

https://www.youtube.com/watch?v=J1fT4ibWUNA

When not changing the layout type, the top two menus remain accessible as can be seen from 0:09 to 0:19. By changing the layout to anything non-default (0:21), the user can one time scroll through "Setup widgets" downwards, but not up back again. The upmost element reachable with roller is now "Top bar" and the roller jumps by further scrolling upwards to bottom-most element "Remove screen". Scrolling the other way, also does not make the Setup widgets and Layout accessible (0:23 - 0:38).

closed 2021-05-27 by raphaelcoeffic

rotorman, 2021-05-27:
Tested with e903a88f. I can confirm that the top elements _Layout_ and _Setup widgets_ are accessible again, so this is nicely solved. Thank you @raphaelcoeffic !

JimB40, 2021-05-27:
Still active in f58317f
https://youtu.be/F_Olv85rEL4

pfeerick, 2021-05-27:
> Still active in [f58317f](https://github.com/EdgeTX/edgetx/commit/f58317f74ed3483e0a1d0384801bcc6c3b62933f)

That's probably because https://github.com/EdgeTX/edgetx/commit/e903a88f24b4a44be828d9f5442483438815ecdb is from about two hours ago and specifically addresses this issue. https://github.com/EdgeTX/edgetx/commit/f58317f74ed3483e0a1d0384801bcc6c3b62933f is an older commit. 😉

JimB40, 2021-05-27:
Okay. Still new to github :) have to read more abt builds not to submit things that are already done

pfeerick, 2021-05-27:
All good... gotta start somewhere ;) And it can be a struggle to keep up sometimes the rate Raphael goes at when he's on a roll ;)

JimB40, 2021-05-27:
Heard from "well informed source" that it's 700% of standard productivity level ;)

commit e903a88f (2021-05-27): Fix focus order after layout choice

Resolves #76