EdgeTX/edgetx #79: Color UI is left in an incorrect state when entering USB mass-storage

raphaelcoeffic, 2021-05-25:
And not properly reloaded after USB mass storage mode is left.

Original description from @pfeerick:
"I can finally report one seemingly new mostly cosmetic bug on current bef88d2 ... go into any menu... radio setup, model setup, etc. plug in the USB, enable storage mode, unplug the USB, and admire... looks like it needs a full screen refresh, and something about navigation is lost/broken - only way to go back seems to via the touch screen, rather than RTN button."

In fact, we're quite lucky the whole thing does not crash (pure luck).

closed 2021-05-25 by raphaelcoeffic

pfeerick, 2021-05-25:
Ouch... So don't walk under any ladders or cross paths with black cats before this is resolved? :laughing:

raphaelcoeffic, 2021-05-25:
> Ouch... So don't walk under any ladders or cross paths with black cats before this is resolved? 😆

In fact it is resolved ;-) See 945fc95.

commit 945fc951 (2021-05-25): Properly reload color UI on USB mass-storage

Resolves #79