EdgeTX/edgetx #83: How about simplifying and improving the API for model settings

lshems, 2021-05-25:
I have a proposal for an, in my opinion and based on my experience, better way of updating/retrieving model settings. We can even start introducing it without killing the legacy code, but in the end nobody will use that anymore if this is going to work as intended (we can keep legacy in for a long time, I assume).

Basic idea is to create a combination of the get and the set functionality in one function, and using a referenced table passed to that function.

Instead of using model.getObject(id) that returns a table, and model.setObject(id,table) that will set the values, I propose to use model.object(id,table) where the table is passed by reference, and returned as a result as well.

- The keys-values provided in the table will update the model settings to the new values, if within range.
- The key-values that could not be updated on the model, will be populated with the current values of the model.
- Any missing key-values will be added with the current model values.
- A new table is created if there was no table passed.
- The table in the arguments passed to the function will also be returned as a function result to allow quick population of settings arrays in the init of any script. 

An extensive explanation with pseudo code examples is given here: https://www.rcgroups.com/forums/showthread.php?3903011-How-I-would-set-up-the-EdgeTX-LUA-API#post47195415

lshems, 2021-05-25:
MMM, whats happening here. A little bit of attention to my blog??

[image]

jmil-dev, 2021-05-26:
Is there an open question here around using LUA to do the config of the radio/models?

lshems, 2021-05-26:
But all the lua model functions do exactly that? Big part of the current API, and cumbersome on a lot of points.

My suggestion is to simplify the existing approach.

And yes, this is very very useful for supporting complex models with custom GUI's.