EdgeTX/edgetx #85: Get rid of the return {run=run, background=background, init=init} closing line on any script

lshems, 2021-05-25:
Can we do that?

- it is confusing, not necessary and scares people off
- make any code run in the current RUN mode, only when the screen is visible. Any code to be run in the background can be defined in a background function. No need to return it as such, just convention on the name. Or even an API function that passes code to a background routine.

lshems, 2021-05-26:
Proposal:
--use the correct way of passing parameters to a chunck
--options would be referencing a table with at least the following keys
--options.event: is what it is now, some way of informing the script of the events that took place since the last pass
--options.state: would indicate if this function call is executed in background, init or run state
-- lets add the focus state as well for instance. It will indicate that this fucntion is called when the screen is active AND has focus. Very usefull for widgets
--options.LCD: would allow to pass an array with the absolute screen coordinates of the left bottom corner in which may be drawn, and the width and height of the screen in pixels available, and the font width and height in pixels for every font size. To be elaborated, but just shown to indicate how powerfull this setup may be in being universal for widgets AND scripts alike.

`local options = ...     --the option parameters will be passed to the script from the LUA API

if options.state = "init" then
  do --something
end

if options.state = "run" then --run will be true if the screen is active
  if options.event then
    do --something based on options.event
  else
    do --something not based on the event, but still when the user is seeing the output
  end
end

if options.state = "background" then --
  if options.event then
    do --why not allow some events to even trigger something in background mode
  else
    do
  end
end

return errorMessage --returning anything at all will stop the script from executing.
`

Pseudocode on the radio
`local options
local userScript = loadfile(scriptNameSetupInRadioGUI)
local errorCode

while errorCode == nil do
  updateOptions(option)--populate the options
  errorCode = userscript(Options)
end

print("errorCode",errorCode)`

lshems, 2021-05-26:
And since the options table is passed 'by reference', you could use it also to return and update those values if neccesary.

Think about updating the options.event to nil for example if you want to kill the events from your script. No extra function calls needed.

Or change the LCD values that where passed, and the Radio will know that in its API management system, and can adapt the output ranges.

rotorman, 2021-05-26:
use three of \` and a line break to start code and again three \` on a newline to end a code block. Like this:
```
code block here
```

lshems, 2021-05-26:
> use three of ``` and a line break to start code-block and again three ``` on a newline to end a code block. Like this:
> 
> ```
> code block here
> ```

Thanks!! updated the post for readability

lshems, 2021-05-26:
With the above approach, we need to discuss what we want with global variables and environments. Currently, telemetry, function and mixer scripts share the same environment. So any variable declared globally from within the scripts loaded are accessale for all.

This allows for data exchange between functions, but this can never be done in a secure way. In other words, if I use two scripts from one author that both use globals for this purpose, and one script from another author that by accident uses the same names for any globals, things are about to go wrong.

Also, especially with new programmers, loading the same script on two functions for instance, with the objective to have two instances running next to each other, will go wrong when the author by accident didn't create his variabels as locals, but as globals. 

(we see this already going wrong terribly wrong with widgets by the way).

The simplest way would be load all scripts with their own environment, so that they are all and every one of them running in their sandbox.

This would also mean that for the above example code, any variables that need to be persistant between function calls simple need to be globals. They will be available ONLY to the script.

Currently, the widget environment as a whole is already separated from the other script environments, and SDcard scripts are run isolated anyway, so no issues there.

That leaves us with the question if and how we want to allow for interchange of data between functions.

lshems, 2021-05-26:
To allow legacy scripts, scripters can use the following conversion script. They give it a new name, and have it load the old script.
The scriptname without path and extension needs to be in the options

I suggest to create a new folder for the userscripts, so not Scripts/telemetry. Just /Scripts would be fine.

so they put this script in the SCRIPTS folder with the same name as the legacy script in the SCRIPTS/telemetry folder.

```
local options = ...     --the option parameters will be passed to the script from the LUA API

-- legacyScript is to be a global to be persistent between calls!
if options and not legacyScript then 
  local errorMessage
  errorMessage,legacyScript = pcall(loadfile("SCRIPTS/Telemetry/" .. options.scriptName .. ".lua")) --load and execute in a protected call
  if errorMessage~=nil then
    return errorMessage
  end
  return pcall(legacyScript.init) --initiate, only once.
end

return pcall(legacyScript[options.state],options.state=="run" and options.event)
```

olliw42, 2021-05-26:
frankly, i dont see any issue with how it is and i actually like being able to use my own names for the functions. i fail to see any value in the proposal, both from lua scripting as well as from spending time point of views. just my 2 cents
this does not mean that it might not worthehile to reduce the number of lua script types or their returns

lshems, 2021-05-26:
The proposal is not worthwhile for me, nor you. If you know your way around in LUA, the current implementation isn't blocking at all.

It is meant to lower the threshold and make lua scripting a doable thing for everyone that can write a basic program.

First step would be to allow a script to not return an array. If no array of functions is returned, it would keep being executing as a function script on every cycle, and just stopped if something else than a table is returned. This would allow already most of what I propose, and allow for a smooth transition to a new approach.

lshems, 2021-06-04:
ok, since Jesper has been working to have LUA run without neccesity to limit time between runs, there is no need anymore to create an API based on runs.
To keep things simple, we should just ensure a few things to get things MUCH much easier for scripting:
1) create a getEvent() function that will return the latest event.
--    in future, buffering of events might be an option to ensure execution even if script is lagging
2) create a lcd.active() function
--    to replace the current run(event) functionality that allows to make complex code for displaying only execute when displaying. just check at start of the display routines if they need ot be executed.
3) have the script executed also if no return array is received.

-- the general format of a function would be something like the following, that would still allow for the return of the {run=run, background=background,init=init} that would still get executed periodically after the script is loaded. However, if the script doesn't return that array, it should still need to get loaded and allow to run indefinitely on the internal loops:

```
local options = ...

--options would be something like {lcd_w = 212, lcd_h=64, color=true, ????}

local x, y, z

function A()
  x=x+1
  return 1
end

local endcondition

loop until endcondition = true
  local y
  if getEvent()==VIRTUAL_PLUS
    y=A()
  end
  if lcd.active() then
    lcd.drawNumber(1,1,x)
    lcd.drawNumber(10,1,y)
  end
end

return something
```

raphaelcoeffic, 2021-06-05:
@lshems thx for your feedback! I'm deeply sorry to have the time right now to properly react to this. Let's continue the discussion after 2.4.0.

lshems, 2021-06-05:
No problem. With the lua 'noloop' pull from Jesper I can already do some
testing ...