EdgeTX/edgetx #86: Make the LUA PRINT command working on a dedicated screen

lshems, 2021-05-25:
Make the PRINT command working on a dedicated screen

- People are thrilled when they can just make a 'HelloWorld.lua', containing the line "print("Hello world")" and have it executed.

After every print command shift the lines down one row, and keep them in the 'print buffer'.
Every line that 'falls off the screen" get's removed from the buffer.

raphaelcoeffic, 2021-05-26:
@lshems so basically like a "logging" window, right?

JimB40, 2021-05-26:
As a quick solution redirecting LUA print output to any file would be really great dev improvement. Now we have to debug using different code than in Companion. If companion supports firmware :) 
simple print with file output would be such a relief cause lcd error formating is kind of disaster. I'm using xpcall but with more data browsing on lcd is cumbersome.

lshems, 2021-05-26:
changed the title, as it seemd confusing. I suggest to open a new issue for the io.stderr:write(message) request.

lshems, 2021-05-26:
logged a new suggestion #90