EdgeTX/edgetx #87: Bug - Emergency Mode if power off whilst 'failsafe not set' screen active

pfeerick, 2021-05-26:
RM TX16S, https://github.com/EdgeTX/edgetx/commit/945fc9510ae451325803b6d2c1ba2ec15b014079

Whilst testing https://github.com/EdgeTX/edgetx/issues/74, I made the mistake of trying to power off the unit whilst the 'failsafe warning: failsafe not set' screen was showing (internal multimod set to frsky, receiver not powered). The shutdown delay screen shows, but before any segments start to show, the screen goes back, then the transmitter goes into emergency mode.

raphaelcoeffic, 2021-05-26:
@pfeerick this should be fixed now with 038b8364f685de9. Can you please re-check?

closed 2021-05-26 by raphaelcoeffic

pfeerick, 2021-05-27:
Confirmed. :) Issue is gone post https://github.com/EdgeTX/edgetx/commit/5960d0aac2c8558073268ee762b9705b2a0590ed

PR #92 (2021-05-26): Moved MPM failsafe check to the UI task
Resolves #87
commit 5960d0aa (2021-05-26): Moved MPM failsafe check to the UI task

Resolves #87