EdgeTX/edgetx #89: No TopBar Icons

JimB40, 2021-05-26:
Hi,

I just have given EdgeTX initial run.
Tried builds tx16s-9691cc9 and tx16s-f97843c from 25th of May

1. Went to https://downloads.open-tx.org/2.4/nightlies/sdcard/ to get updated SD card content
- all files are named V0026
- there is no file for RM TX16S 
Downloaded  the one for Jumper T16 as closest one

2. After power up with EgdeTX I can't see single icon on Top Bar including Logo. 
EdgeTX seems to work as expected only icons are not displayed

3. There is no way to change Theme (maybe intentional) so assumed Default is used, Then checked /THEMES/DEFAULT/ folder in downloaded zip and there are only 4 files (including two mask_****.png)
Copied missing files from  /THEMES/DARKBLUE/  but still TopBar icons are not visible.

Bug or I'm doing something wrong?

[image]
[image]
[image]

raphaelcoeffic, 2021-05-26:
@JimB40 you need to copy the EdgeTX theme located here on the repo (`radio/sdcard/horus/EdgeTX`) into `THEMES` on the SD card. That is necessary until these simple bitmaps are integrated into the firmware itself (see #15). Concerning the theme choice, it is indeed not yet implemented, but will be soon.

JimB40, 2021-05-26:
Thank you Raphael. Now works properly. :) 

I've also tried crossfire.lua and AgentLite. None of them works. Fired up my LUA CRSF packet sniffer and seems like there is no response from microTX  (0x28 frame). 
Any changes were done in LUA crsfTelemetryPush() & crsfTelemetryPop()? Data filtering? Some frames (mainly broadcast) are not passed to Pop function in OpenTX, instead they are changed into OpenTX telemetry fields.  
[image]

raphaelcoeffic, 2021-05-26:
> Any changes were done in LUA crsfTelemetryPush() & crsfTelemetryPop()? Data filtering? Some frames (mainly broadcast) are not passed to Pop function in OpenTX, instead they are changed into OpenTX telemetry fields.

I don't think so. It has not been investigated. If that persists, or you think there is an issue here, please make a separate one.

JimB40, 2021-05-26:
Okay. During radio and model setup I've spotted few minor UI bugs or quirks. Want them in separate issues? Don't want to flood :)

raphaelcoeffic, 2021-05-26:
> Want them in separate issues? Don't want to flood :)

You can make a list of "minor UI bugs or quircks" and place that into an issue, no problem.

closed 2021-05-26 by raphaelcoeffic