EdgeTX/edgetx #96: LUA improvements/feature requests

lshems, 2021-05-26:
I know I have posted a lot of proposals for changes. I do not want to push for anything. There are perhaps other objectives of this fork. If so, I fully respect that. I posted this feature request to try and keep track of the proposals I made, and collect coordinated feedback, since a lot of my request are related and interdependent.
I will not post any more feature requests for LUA as separate requests, but will add them in this feature request instead, until there is feedback and agreement on the feasibility and do-ability of these changes.

Personally, I would love to see this common goal supported:

**_Create a community of RC enthousiasts that actively develops and shares scripts that are useful for the community itself._**

Which I think is not possible with some changes in the way LUA is integrated in EdgeTX.

I do appreciate the any feedback, positive or negative.

Guido.
([see my blog on rcgroups as well](https://www.rcgroups.com/forums/showthread.php?3903011-How-I-would-set-up-the-EdgeTX-LUA-API))

Current feature requests:
https://github.com/EdgeTX/edgetx/issues/86
https://github.com/EdgeTX/edgetx/issues/85
https://github.com/EdgeTX/edgetx/issues/84
https://github.com/EdgeTX/edgetx/issues/83

CoderElectronics, 2021-05-26:
Let me take a look at those changes, I can say a few things like the Model API will be difficult to successfully ratify due to the fact that many scripts rely on it.

lshems, 2021-05-26:
The proposal for the model API allows backward compatibility.