EdgeTX/edgetx #102: Select stick mode and default channel order on storage reset

raphaelcoeffic, 2021-05-27:
This feature should allow the user to:
- select the stick mode
- select default channel order

Whenever the radio file is reset (or not existing). Ideally, it should be possible to wipe everything clean ("factory reset") and re-trigger this from the radio settings menu.

rotorman, 2022-01-02:
If a user is using extra channels, e.g. on TX16S the EX1 to EX4 (also digital SI & SJ), then current first power-on experience with no configuration on SD card, is to power-on, run the calib once. Then, a user can first enable the EX1 to EX4 (plus SI or SJ) and need to RE-RUN the calib. It would make sense to be able to select on a clean system e.g. "Advanced mode" and select which extra channels are used/wired-up, before the analog calibration procedure is triggered.

raphaelcoeffic, 2022-01-02:
@rotorman yeah, we could have a button that would take you to the hardware settings first.

also, we should really have the internal module section be part of this.

rotorman, 2022-01-04:
By reading RCG, lots of users are not ware on T16/TX16S radios that 6-pos is an analog channel and needs to be actuated correctly also during analog calibration. It might be a good idea to hint this during calib screen as well, on radios that come with analog 6-Pos (like Jumper T16/T18, RM/Eachine TX16S).