EdgeTX/edgetx #104: EM on power down

JimB40, 2021-05-27:
Goes to EM every single time. Same with previous builds.
https://youtu.be/D9g3GjSMScE

**General information**
EgdeTX version f58317f
RM TX16S

rotorman, 2021-05-27:
This seems to happen when at least one LUA widget is active on any page. Can you please test if the EM comes as well, when none of your pages have LUA widgets included? On mine, then there is no EM screen on shut-down anymore.

JimB40, 2021-05-27:
Tested. Removed all LUA widgets even those on topbar.
Still EM every time

rotorman, 2021-05-29:
PR #113 solves the issue for me.
@JimB40 how does it work for you? If positively, we could close this issue?

JimB40, 2021-05-29:
Little confused from which version of FW should get to test.
Downloaded today tx16s-a972353.bin from artifact but file for tx16s has 134MB, so I was little hesistant to write it :)

rotorman, 2021-05-29:
@JimB40 Try under Artifacts here: https://github.com/EdgeTX/edgetx/actions/runs/888034651

JimB40, 2021-05-29:
Seems that is solved

https://youtu.be/1wN3cvuvAnk

Chapeau bas 😀

rotorman, 2021-05-29:
Awesome @JimB40 and thank you for the video.
Can we close the issue?

JimB40, 2021-05-29:
Think so. I'll give more intensive test run in afternoon. Will comment here if I encouter any problems.

closed 2021-05-29 by raphaelcoeffic