EdgeTX/edgetx #105: Crossfire on internal module

DeepnineRC, 2021-05-27:
Is there a way to remove the stock frsky internal module and replace it with a crossfire or any other external module?

Background is I am not using the frsky ixjt for years, but I am switching from crossfire to elrs all the time.

Nice to have would be a 2G4 Elrs internal on the stock antenna.

I found INTERNAL_PXX1 INTERNAL_PXX2 INTERNAL_MULTI INTERNAL_PPM

I ready for soldering my old x9d and my x12s if someone can do the programming.

Thank you

FractalEngineer, 2021-11-13:
love it, been waiting on this for a while to make an internal ELRS module for X-lite!

raphaelcoeffic, 2021-11-13:
> love it, been waiting on this for a while to make an internal ELRS module for X-lite!

Work has started on this, stay tuned 🤓

pfeerick, 2021-11-28:
Closed by #1117

You may find #1039 may be of interest also ;)

closed 2021-11-28 by pfeerick