EdgeTX/edgetx #106: Full screen LUA widget misbehave when touch menu used

JimB40, 2021-05-27:
Using touch menu to go into Full screen mode makes script & menu unresponsive

**To Reproduce**
Steps to reproduce the behavior:
1. Double click on full screen LUA widget
2. Tap on 'Full screen'

**Screenshots**
https://youtu.be/wNYlzZG04zA

**General information**
EgdeTX version f58317f
RM TX16S

raphaelcoeffic, 2021-05-27:
@JimB40 Thx for all the testing! You will be rewarded with a better software :-)

JimB40, 2021-05-27:
I will get reward when my LUAs will have better UX. (I'm behind Agent Lite dev) 
Worked few years coding in PHP then in Soft/HW testing, then UI/UX design so got some experience. 
I know how to use PS and mockup things.
Now photographing but old love never dies ;) 
Not much C++ knowledge but I can contribute in UI/UX, testing or documentation. 
Was quite frustrated that lot of LUA improvements were stalled or rejected in OpenTX for more than 1,5 year. Even someone prepared code to pull and showed reasonable arguments to include it.
BTW really good work guys. Thank you.

raphaelcoeffic, 2021-05-27:
@JimB40 by the way, I guess you have tried as advised to exit full-screen mode with [RTN] pressed long, right?

JimB40, 2021-05-27:
Yes I have.
As far I tested it goes back properly with [RTN long] press. Only touch menu stays displayed forever. 
There is no clear UI confirmation after [RTN long] press that widget works now old-way (key's events processed by EgdeTX) so not easy to tests.
Really think FullScreen widget should initiate in this new mode with key's events processed by script. Then after [RTN long] is pressed screen should be dimmed with info that we can now switch between main views or go back to Full screen mode pressing [ENTER] or double tapping. 
As I mentioned before for Full screen widget menu (Full screen & Widget settings) or this red frame to select widget is not really needed as it only confuses because there is only one widget to select.
  
I think code is for every widget's layout so this would need to code exception.

raphaelcoeffic, 2021-05-31:
@JimB40 i had hard times reproducing it. Could please re-test on latest code?

raphaelcoeffic, 2021-05-31:
> Really think FullScreen widget should initiate in this new mode with key's events processed by script. Then after [RTN long] is pressed screen should be dimmed with info that we can now switch between main views or go back to Full screen mode pressing [ENTER] or double tapping.

I really want to avoid going to a main view page and being “automatically” (I.e., without user interaction) in full screen mode.

I agree though with the message to be displayed after changing back to normal mode, similar to when switching to full-screen.

JimB40, 2021-05-31:
From UI/UX point of view when you swipe to Yappu/iNav/AL or whatever you want see app running. No additional clicks touches, looking how to select red frame etc. Are there any OS runtime obsctacles to archevie it?
Or maybe we go back to this 'special' app fullscren widget concept where the only config is to point to app's main lua file?

raphaelcoeffic, 2021-05-31:
> Are there any OS runtime obsctacles to archevie it?

no, apart from the fact that it feels weird when get there with swipe/page key, but cannot get the other direction the same way, because the apps takes all inputs.

JimB40, 2021-05-31:
Understand. Sounds reasonable when user has few apps.
Then I will insist on special full-screen-app widget mode. Initial state is little dimmed and you can swipe further or back then tap or [ENTER] enables passing key-events to app and [RTN long] goes back to dimmed-swipe-posible state with key-events passed to edge. No pop-up menu. :)

raphaelcoeffic, 2021-06-02:
@JimB40 I think that's ok, but we should have some sort of "story book", basically showing picture-by-picture how it would look like, so it is easier for me and others to picture how it would be once implemented.

JimB40, 2021-06-02:
Workin on that.
This high level subject so functional solutions' questions will pop up here and there. I will create issue for that even this it is not 'issue' at all :)

raphaelcoeffic, 2021-06-06:
@JimB40 concerning the original issue here, I cannot reproduce it any more. Can I please close it?

JimB40, 2021-06-08:
Well I can every single time :)
https://youtu.be/v6N9apHobK4

raphaelcoeffic, 2021-06-08:
> Well I can every single time :)
> https://youtu.be/v6N9apHobK4

Damn! I really need to catch this thing!

raphaelcoeffic, 2021-06-08:
@JimB40 thx for the demonstration! Now I have it reproduced here. It should not take long.

lshems, 2021-06-08:
> Understand. Sounds reasonable when user has few apps.
> Then I will insist on special full-screen-app widget mode. Initial state is little dimmed and you can swipe further or back then tap or [ENTER] enables passing key-events to app and [RTN long] goes back to dimmed-swipe-posible state with key-events passed to edge. No pop-up menu. :)
> 
> With every other widget size pop-up menu with "full screen" is okay. So LUA app can have let's call it 'mini version' kind of overview or just logo. Then when you go full-screen it works in the same way as described before. After [RTN long] it goes to back to 'mini-version' and view with more than one widget is displayed.
> If apps will have this mini-state developed well it will be possible to have kind of quick-list or shotcut-list. No need to swipe between views in such a UI setup
> 
> Sounds good?

That sounds perfect. All my apps are already scalable to any size. 

Nice!

lshems, 2021-06-08:
That would allow to skip my main page and have all stuff in widgets ready to pop up. Cool 😎😎😎

raphaelcoeffic, 2021-06-08:
> That would allow to skip my main page and have all stuff in widgets ready to pop up. Cool 😎😎😎

Gentlemen, this discussion is a bit off-topic in this issue. Would you discussing this is another issue? 😅
I will have to close this one as I found the original issue.

closed 2021-06-08 by raphaelcoeffic

lshems, 2021-06-08:
This edge thing is exiting and while writing I wonder how long I can hold off. 

Lol

pfeerick, 2021-06-09:
Feel free to take it over to https://github.com/EdgeTX/edgetx/discussions and keep the ideas flowing ;)

JimB40, 2021-06-10:
Yes please. leave bugs issues just for debugging. otherwise once issue is closed side information is 'gone' and harder to find

commit 25263b16 (2021-06-08): Forward touch event to child windows even for widgets

This is the case when the menu is started whereby the widget takes the full screen.

Resolves #106