EdgeTX/edgetx #114: T16: TBSAgentLite.lua Does Not Run

vileperson, 2021-05-28:
**Describe the bug**
TBSAgentLite.lua fails to run on Jumper T16. It simply displays a "graphical glitch" type of screen. I am able to back out of it by long pressing RTN

**To Reproduce**
Steps to reproduce the behavior:
1. Load TBS Agent Light lua script and folder to SD Card per TBS instructions here. https://www.team-blacksheep.com/products/prod:agentx
2. Go to 'SDCARD'
3. Click on 'SCRIPTS', 'TOOLS'
4. Scroll down to 'TBSAgentLite.lua.'
5. Long press enter (scroll wheel), and select EXECUTE.
6. See graphic loading issue

**Expected behavior**
should load the main menu for the Agent Lite script

**Screenshots**
[image]

**General information**
 - t16-64330e3.bin
 - Jumper T16
 - TBS Agent Lite

raphaelcoeffic, 2021-05-30:
@vileperson this will require the source code of the app, which I don't have right now. Need to get TBS approval first....

raphaelcoeffic, 2021-05-31:
Ok, no approval from TBS, which means, it is TBS to fix it ;-)

From what I have seen so far, what is seen on the screen is basically the content of the LUA back-buffer, in which LUA routines draw before it gets copied to the real LCD buffer. This means that the script neither clears nor draws anything at this point.

tbs-trappy, 2021-05-31:
We are looking into this on our end and will release a new revision from Agent Lite as soon as possible. thanks for reporting this bug!

raphaelcoeffic, 2021-05-31:
> We are looking into this on our end and will release a new revision from Agent Lite as soon as possible. thanks for reporting this bug!

I would rather prefer have an explaination as to why it does not work, so I can fix in EdgeTX. The goal here is to have Apps compatible with OTX 2.3 work in EdgeTX without any modifications.

tbs-trappy, 2021-05-31:
> I would rather prefer have an explaination as to why it does not work, so I can fix in EdgeTX. The goal here is to have Apps compatible with OTX 2.3 work in EdgeTX without any modifications.

Yes, will post our findings here. Right now we're just getting ready to reproduce the problem :)

JimB40, 2021-05-31:
I've checked on T16 simulator and no trace of error.
Can you check if you have latest TBS AL build?
Open TBSAgentLite.lua from /SCRIPTS/TOOLS/ folder and chek for this line:

local run = loadScript('loader.lua','bt')('0.91')

vileperson, 2021-05-31:
> I've checked on T16 simulator and no trace of error.
> Can you check if you have latest TBS AL build?
> Open TBSAgentLite.lua from /SCRIPTS/TOOLS/ folder and chek for this line:
> 
> local run = loadScript('loader.lua','bt')('0.91')

Yes, that line is the same in my version. I had pulled the latest AL before reporting the bug, just to be sure.

JimB40, 2021-05-31:
Can you test again against latest build? @raphaelcoeffic pushed lot of LUA improvements lastly.
If bug persist we'll be tracing further.
Btw so far there is another bug that prevents communication LUA<->CRSF module but UI should load without glitch

vileperson, 2021-05-31:
> Can you test again against latest build? @raphaelcoeffic pushed lot of LUA improvements lastly.
> If bug persist we'll be tracing further.
> Btw so far there is another bug that prevents communication LUA<->CRSF module but UI should load without glitch. Don't pay attention to colors as color indexes are under dev.
> [image]

same issue with the latest build...
[image]
 It looks as though the T16 is trying to load the EdgeTX bootloader graphic for some reason, instead of the TBS AL graphic.

raphaelcoeffic, 2021-05-31:
> It looks as though the T16 is trying to load the EdgeTX bootloader graphic for some reason, instead of the TBS AL graphic.

what you see is empty but not wiped screen buffer. It contains the last image produced by LUA content. Try to load another working LUA script before.

vileperson, 2021-05-31:
If I run the regular crossfire lua first, and then try to run agent lite, it displays the crossfire lua screen. So yeah. That's exactly what it's doing... displaying the previously loaded screen.

raphaelcoeffic, 2021-06-15:
Please note that this is not specific to T16, I have the same on the TX16S.

JimB40, 2021-06-15:
okay so in response to LUA:
local bg = lcd.RGB(230,230,230)
local txt = lcd.RGB(0,0,0)

OpenTX returns:
2021-06-15 14:06:48 debug bg: 59196
2021-06-15 14:06:48 debug txt: 0

While EdgeTX
2021-06-15 16:08:07 debug bg: nil
2021-06-15 16:08:07 debug txt: nil
2021-06-15 16:08:07 /SCRIPTS/TOOLS/TBSAGENTLITE/splash.lua:24: bad argument #2 to 'setColor' (number expected, got nil)

raphaelcoeffic, 2021-06-15:
@JimB40 that is easy to fix! Wait a sec!

raphaelcoeffic, 2021-06-15:
seems 60517df3496 fixes it.

closed 2021-06-15 by raphaelcoeffic

vileperson, 2021-06-17:
Fix confirmed with release candidate, here.