EdgeTX/edgetx #116: Virtual Keyboard for inputing text

phileville, 2021-05-28:
**Is your feature request related to a problem? Please describe.**

Currently, in order to enter text anywhere, you have to use the scroll wheel to scroll through the alphabet until you get to the letter you want - as it was before the touch interface was active. This is not very efficient for users with touch screens.

**Describe the solution you'd like**
Ideally, when a user touches a text box, a virtual keyboard would appear and the user would be able to type the text by touching the letters on the virtual keyboard and when finished, press an enter key (or somthing like that) and the virtual keyboard would disappear again. For those that do not use the touch screen, the scroll wheel would just scroll throught the letters on the virtual keyboard (kind of like before) until they select one.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
I think the issue is relatively clear, if not -  Try to change the name of a model and you will see what I mean.

lshems, 2021-05-28:
Create a lua widget for that? Can do for model name. But indeed, would seem logical.

But not only for touch.

Xlite have a very nice  navigation joystick you could use more efficiently than what we have now as well on a virtual keyboard.

raphaelcoeffic, 2021-05-29:
The good thing is: the virtual keyboard is already done/prepared in libopenui, so that we “just need to use it”.

evildronepilot, 2021-07-06:
This is like 50% of the touch screen useful functionality => absolute must-have

raphaelcoeffic, 2021-07-06:
> This is like 50% of the touch screen useful functionality => absolute must-have

No worry, I’m with you.

PR #644 (2021-08-30): Software keyboard support.
Closes #116

- Centered keyboard
- Cleaned up all the magic numbers
- provide visual feedback on TouchStart
- fixes hit boxes to be halfway before the letter and halfway after.  the old logic was at the first pixel of the letter to the right, making key selection haphazard.  
- Fixed the bitmap position of the backspace.
- fixed issue with having edited fields be off screen

most of the work was in libopenui which will have a separate pull request
closed 2021-08-30 by pfeerick