EdgeTX/edgetx #118: Autofill Switch Selection Not Working & Sound Tracks Not Available/Not Working in Drop-down Options

TMac54, 2021-05-28:
**General Information**
Transmitter:  RadioMaster TX16S
EdgeTX version:  using tx16s-564dda3.bin & SD Card contents from 26 May, 5:23 pm

**Describe the bug**

1.  Autofill feature of selecting switches as sources in Mixes Page and Special Functions Page does not work.  Instead, you need to manually scroll through the drop-down menu options to choose the source you wish to use from the drop-down menu.  

2. When attempting to use a sound track in Special functions Page after selecting "Play Track" and then selecting the Value field, only lq.wav appears in drop-down menu (even after I have uploaded a sound track which was working properly with OpenTX, it does not work in EdgeTX)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Special Functions Page (and/or Mixes Page)
2. Scroll to empty SF line and press enter to edit
3. Press enter in Switch field.  Flip switch of choice - nothing happens (need to scroll through list of options to select the one you're interested in using rather than flipping switch and having it "autofill" for you

4. Scroll down to Function field and press Enter
5. Select Play Track
6. Scroll to Value Field and Press Enter
7. Even though other sound tracks are on SD card (I've uploaded custom sound tracks which did and do play as sound tracks using OpenTX) , the only one that is displayed to be selected is lq.wav (unable to select other tracks).  Even when lq.wav is selected, after having set it up on a switch, the sound track does not play

**Expected behavior**
1.  Expected to be able to flip a switch to "autofill' entry field of menu options
2.  Expected all sound tracks loaded on SD card to be available for selection and then to play properly when activated by Special functions switch as source

raphaelcoeffic, 2021-05-29:
Thanks for adding this! It’s on my TODO list for some time now (love the autofill), having an issue is better for tracking progresses.

TMac54, 2021-05-31:
@raphaelcoeffic Thanks for adding it to the 2.4.0 milestone!
Have the sound tracks also been addressed?

raphaelcoeffic, 2021-06-02:
> Have the sound tracks also been addressed?

Thx for creating a second issue, makes more sense ;-)

commit babb7405 (2021-06-06): Added auto-switch / auto-source to colorlcd

Resolves #118
commit e229e852 (2021-06-07): Added auto-switch / auto-source to colorlcd

Resolves #118
PR #155 (2021-06-07): Added auto-switch / auto-source to colorlcd
Resolves #118
commit 162ffdb1 (2021-06-07): Added auto-switch / auto-source to colorlcd

Resolves #118
closed 2021-06-07 by raphaelcoeffic