EdgeTX/edgetx #121: tango 2 support milestone

squshy7, 2021-05-29:
I know work is being made in the background to merge freedomtx code to get tango support into edgetx, but I noticed there is no milestone for either 2.4 or beyond. I'd like to see this out in the open so the community knows it's being worked on, and can monitor the progress on that work.

raphaelcoeffic, 2021-05-29:
@squshy7 As far as I'm informed, the port should come to OpenTx first, as TBS started with OpenTx. We will pick it up then as soon as merged into OpenTx.

squshy7, 2021-05-29:
> @squshy7 As far as I'm informed, the port should come to OpenTx first, as TBS started with OpenTx. We will pick it up then as soon as merged into OpenTx. I have no further information at the moment.

TBS seems to be under the impression that opentx is the one working on this...but this sounds like you heard TBS was working on it? That would not be good

[image]

raphaelcoeffic, 2021-05-29:
No, that’s what I meant: OpenTx is working on finalizing the integration. Last news I had.

squshy7, 2021-05-29:
> No, that’s what I meant: OpenTx is working on finalizing the integration. Last news I had.

ok gotcha, thanks. closing

closed 2021-05-29 by squshy7