EdgeTX/edgetx #125: Main Menu: Scrolling forward makes the red box move forward in strange ways

olliw42, 2021-05-30:
I tried to figure out how to correct this but I failed to understand libopenui, hence raised as an issue:

In the main menu, scrolling to the right does not make the red select box move as expected:

When freshly opened, the red box moves 3 options forward, but then one is trying to move to next, 4th and last visible option, it does not go there but jumps to the 1st position in the menu and with it the 4th option jumps to the 1st position (such that the 4th option is selected). This repeats itself with going on. Only then the very last option is reached the red box goes to the 4th position, as it should be. With a further scroll it jumps to the 1st position selecting the 1st option (as it should be).

In contrast, when scrolling to the left, the behavior is as expected:

Here the red box moves to the left and if it is in 1st position it stays there and the menu options are moved one by one. Then the 1st option is reached, it jumps to the 4th position restarting from the end of the menu. That's how it should be, and that's also how the selection in all other menu pages work.

raphaelcoeffic, 2021-05-30:
@olliw42 can you please make screenshots that describe the single steps so I can better understand?

olliw42, 2021-05-30:
I know, I know ... but it is actually not so easy to capture descriptively by screenshot ... it would need a video ...

I however think that by trying it along with reading the text should make it clear what the issue, so just do this:
* from the main page press enter to open the main view
* scroll 8 times to the right
* the scroll backwards 100 times

rotorman, 2021-05-30:
The issue is visible in the following (older) video at 0:05 mark, when I try to scroll to the right:
https://www.youtube.com/watch?v=r7CDq8q5xPo?t=4

olliw42, 2021-05-30:
no, that's not the issue I'm talking about
what you see I also often see with other menus, namely that after the menu had been opened the first scroll to the right/down does actually result in a move left/up/to the end
that's an issue too, but not the one here :)

raphaelcoeffic, 2021-05-30:
> what you see I also often see with other menus, namely that after the menu had been opened the first scroll to the right/down does actually result in a move left/up/to the end

I'm afraid this is hardware issue by design. The roller on T16 & derivates is a catastrophe. We've done our best to make it work, but it is still not perfect.

You can test this is actually working by launching the "simu" program and actually trying it there out. You will see that it works flawlessly. On a X10, there is no such issue. The roller always works perfectly. @Risto suggested some hardware changes to make it actually work properly.

olliw42, 2021-05-30:
> I'm afraid this is hardware issue by design. The roller on T16 & derivates is a catastrophe. We've done our best to make it work, but it is still not perfect.

I'm happy to accept that ... BUT: 

this is NOT the issue I'm reporting here !!! Let's not get distracted from the main theme :)

rotorman, 2021-05-30:
@olliw42 I made a video with the clicks you mentioned, but cannot unfortunately see the issue:
https://www.youtube.com/watch?v=YCRj8GnREf0
Could you please make a video or if the issue is seen in the video above, say at which time it occurs. Thanks!

olliw42, 2021-05-30:
THX for the video

it DOES show exactly the issue !!

look at 0.14, where you do the 3th roll to right ... the red box does NOT move to the 4th position/option, here "Screen Setup", but instead the red box jumps back to the 1st position and along with it do the options, so that the "Screen Setup" option is now at the 1st position (and selected by the red box as it should) !!!  
This happens again at 0.17, after the 6th roll to right, there the red box does NOT move to the 4th position but instead stays at the 3rd position!

All this does not happen then you scroll to the left.

rotorman, 2021-05-30:
Aah now I see what you mean - scrolling is not uniform between scrolling to the right and scrolling to the left. Very well spotted.

Although the jump at 0:14 did not disturb me yet (as the cursor remains selected on the menu item and does not wildly jump to smth. other), I see now, that it might indeed be more uniform if the red selection box to stay at rightmost position, when scrolling to the right, the icons would change (shift to left) and only on switch from a further scroll from _About EdgeTX_ towards right, there would be a position change of the red selection box. Is this in your vain?

olliw42, 2021-05-30:
that's a very long sentence, but I think the answer is: yes

:)

as said, it's just the behavior when scrolling to the left, and for all other menus I'm aware of

raphaelcoeffic, 2021-05-30:
Ok, got it! 😁

pfeerick, 2021-06-01:
he he... I wasn't going to mention that one yet... I figured it was 2.4.1 material... I found it quite disconcerting until I realised what was happening (right vs left).

raphaelcoeffic, 2021-06-02:
No Problem with changing that behaviour, but I would be very thankful for a precise description (algorithm?) of how you think it should work. We can then try it out, and give feedback, so we know which option feels better for everyone. Otherwise I can start studying other apps and try to mimic that behaviour.

pfeerick, 2021-06-02:
I see two primary UI options (would love to hear more):

1.  If the selection box is on the left side (default position when you open the menu), and you scroll right, it moves right until it hits the right most icon position (i.e. fourth icon on TX16 - I'm assuming other screen orientations may change the layout) and then the icons start moving, not the selection box. If it's on the right side, and you scroll left, the selection box moves to the left-most position, and if you keep scrolling, it stays put, and the icons start moving. 
1a) You can scroll to the left, and it stops at the first icon, and when you scroll to the right, it stops at the last icon. i.e. It does not "wrap around". 
1b) You can scroll to the left, and keep going beyond the first icon, and the same when scrolling right. i.e. It does "wrap around". 

2. Make it three icons wide instead of four... which then allows you to put the selection box in the middle... then the box doesn't move, and icons scroll by, left or right. I think this lends itself well to having wrap-around behaviour.

olliw42, 2021-06-02:
> I would be very thankful for a precise description (algorithm?) 

what about: It should just behave like for scrolling backwards, or in all other menus, as mentioned before ???

raphaelcoeffic, 2021-06-02:
@JimB40 please come to my rescue ;-)

JimB40, 2021-06-02:
Okay definitely red frame should go to 4th posion scrolling right and to 1st scrolling left before anything happens with icons.

When red frame is in edge position we have two possibilities
a. Start to scroll icons leaving frame in edge position
b. Flip page (like with swipe) and put frame on 1st position when scrolling left or on 4th position when scrolling left

This is to be tested but my experience tells me option b may be better. Constantly switching icons that are similiar give little confusion. 

Last but not least endless scroll. Question is if 
- being on 4th posion on 2nd page switch to first page scrolling right or 
being on 1st position on 1st page switch to second page scrolling left

We should test. Again my experience says: No. scrolling should stop on 1st (left scrolling) and 8th (right scolling) icon.
Roller nav is fast and endless scroll gives confusion sometimes.

raphaelcoeffic, 2021-06-02:
> Roller nav is fast and endless scroll gives confusion sometimes.

I was thinking about the same. For the other question, I would tend to option a, I think, which is the behavior currently scrolling to the left.

rotorman, 2021-06-02:
My preference is also to the way scrolling to the left is currently implemented. Even if I did not notice the issue previously at all, before @olliw42 pointed it out to me, as I am now aware of it, I do notice it more prominently and see the icon jump negatively (at 15 sec. mark https://www.youtube.com/watch?v=YCRj8GnREf0).

JimB40, 2021-06-03:
To be considered. Although any fancy UI animations (I like them too 😀) give Wow effect, they have to be used with caution.
1. They should not slow down UI response
2. Used if really needed. After using tool for longer time wow effect disappears. We just quickly want to do a task and anything slowing down this process creates irritation.

pfeerick, 2021-06-03:
My preference (which I deliberately didn't express before) if going the four icon route would be to not have endless scrolling, as the menu is so short that it is pointless and would just confuse my 'lil homer mind. Additionally, there should be no jumping of the selection box from one side to the other... it doesn't seem sane UX design that that the selection box would jump in the _opposite_ direction to what I'm scrolling. ;)

JimB40, 2021-06-04:
Just implement imho. 
2.4 will serve as solutions' test field to gather input.
Already working on UI with 'no more than 3 clicks to get shit done'.
Still we need working and debugged solution in 3 weeks :)

pfeerick, 2021-11-11:
Please don't just close issues without comment, and also this issue has _not_ been resolved - just the selection box has changed colour.

JimB40, 2021-11-11:
Sorry. Was doing UX/UI issues review. Testing before pressing close. So SD card browser works okay
Havent noticed it's about quick menu too.

pfeerick, 2022-07-01:
Closing as scrolling is now consistent after LVGL and menu redesign.

closed 2022-07-01 by pfeerick