EdgeTX/edgetx #128: LUA Virtual Exit event missing?

lshems, 2021-05-30:
How do you guys manage the deselection of an item selected for editing?

I used to use EVT_EXIT_BRK, but it isn't triggered (anymore) on my XD09. 

I get a 97, but is this still linked to a key constant?

Don't want to hardcode, as this is again one of the magic numbers that may change between releases and vary between radios....
??

raphaelcoeffic, 2021-05-30:
@lshems you should use the virtual constants, so basically `EVT_VIRTUAL_EXIT`. I think `EVT_EXIT_BREAK` is not defined at all.

raphaelcoeffic, 2021-05-30:
> Don't want to hardcode, as this is again one of the magic numbers that may change between releases and vary between radios....

This should be checked and fixed, that's right. But I can assure you that the virtual `EXIT` event works.

lshems, 2021-05-30:
Unfortunately the event on exit gives 97 first, and in my scripts I never arrive at the 33. The 33 does come indeed after the 97 in a simple one time test script. 

Seems my script gets aborted before it reaches the 33. OK, i'll search further on my side.

closed 2021-05-30 by lshems